trezor/trezor-mcu #230: When reset is unfinished, Trezor has the new seed without the user reading it

runn1ng, 2017-10-25:
If user disconnects Trezor during the seed display, the seed is written on the device memory, but the user cannot read it anymore.

It can happen for example with faulty connection. User has wrong cable, doesn't see all his seed words during init, but trezor acts like he already saw it and starts to give him public keys etc. from the unseen seed.

saleemrashid, 2017-10-25:
This is the culprit https://github.com/trezor/trezor-mcu/blob/v1.5.2/firmware/reset.c#L121

saleemrashid, 2017-10-25:
Untested patch

```diff
diff --git a/firmware/reset.c b/firmware/reset.c
index 8217d27..6e88e9d 100644
--- a/firmware/reset.c
+++ b/firmware/reset.c
@@ -118,7 +118,10 @@ void reset_backup(bool separated)
 
 	storage.has_needs_backup = true;
 	storage.needs_backup = false;
-	storage_commit();
+
+	if (separated) {
+		storage_commit();
+	}
 
 	for (int pass = 0; pass < 2; pass++) {
 		int i = 0, word_pos = 1;
```

runn1ng, 2017-10-25:
Why `if (separated)`?-

The issue is simmilar there - when the separated backing up fails because of wrong hardware/transport/..., you don't want to never show it again.

On the other hand, if the storage_commit is always on the end even there, user can "trick" the Trezor with disconnecting and have "unlimited backups" (because he can disconnect after the first "round" of seed words). :)

saleemrashid, 2017-10-25:
@runn1ng We discussed that, but @prusnak decided this was the best solution.

I still maintain that the latter solution is better as we should allow the user to compromise their security if they really want to. We've made it clear enough that it's insecure and there are lots of other ways they could mess up. But I've made my patch match @prusnak's intentions.

We could always reopen dialogue about this.

runn1ng, 2017-10-25:
Oh OK, I missed that debate. OK then

PR #241 (2017-11-14): Commit storage after initialization
Fixes #230
commit e3460b9f (2017-11-14): reset: Call storage_commit after initialization

Fixes #230
closed 2017-11-14 by prusnak