trezor/trezor-mcu #231: OP_RETURN output is not displayed on device

runn1ng, 2017-10-26:
When making an OP_RETURN output, the info about the output is not displayed on device.

I am not sure whether this is a bug or a feature. The output has 0 value, but user is basically signing the data with his private key. So the data should be displayed.

saleemrashid, 2017-10-26:
@jhoenicke and I discussed this briefly. I think it would be good to display it as, for example, people have been known to send Counterparty or OmniLayer tokens to their TREZOR, but malware could steal them during another transaction.

Not sure how useful displaying the actual data is though or if we should just display that there is data.

prusnak, 2017-11-15:
Implemented in https://github.com/trezor/trezor-mcu/commit/54659d49d85024898d2fd31c4db9da12b4ae96f9

closed 2017-11-15 by prusnak