trezor/trezor-mcu #232: Tx fee estimation can be wrong with op_return

runn1ng, 2017-10-26:
When transaction has OP_RETURN outputs, the calculation of fee can be wrong

https://github.com/trezor/trezor-mcu/blob/d006ef6bf7edfa2334fa4d03f004aa09d072526a/firmware/signing.c#L594

OP_RETURN output scripts have variable length (from I think 2 up to 41 bytes), this calculation asumes fixed output length.

I am not sure if this can be used for some attack or not

runn1ng, 2017-10-26:
Perhaps somehow chained with #231 ?

runn1ng, 2017-10-26:
OP_RETURN is not displayed, so attacker can modify the transaction to include thousands of OP_RETURNs and the user won't detect it, and will force it to pay larger fees?

saleemrashid, 2017-10-26:
@runn1ng You can have no more than one `OP_RETURN` output per transaction and each `OP_RETURN` output is limited to 80 bytes. Therefore, the most that an attacker could add to the transaction would be 90 bytes (8 bytes for amount, 1 byte for script length, 1 byte for `OP_RETURN`, 80 bytes of data)

saleemrashid, 2017-10-26:
That would mean that the estimated transaction size would be off by no more than 55 bytes (which seems acceptable to me). Also, we're underestimating the size rather than overestimating, so we would calculate the fee rate as higher rather than lower and be more likely to warn the user.

runn1ng, 2017-10-26:
Oh, there is just one op-return per tx, I forgot that

runn1ng, 2017-10-26:
I am closing this, since this seems fine

runn1ng, 2017-10-26:
> You can have no more than one OP_RETURN output per transaction and each OP_RETURN output is limited to 80 bytes

Hm, that only applies to what is standard, but Trezor still lets you make the transaction, right. It won't relay, but it stil valid transaction, right?

saleemrashid, 2017-10-26:
@runn1ng While that's true, the below still applies so I don't think this is an issue

> Also, we're underestimating the size rather than overestimating, so we would calculate the fee rate as higher rather than lower and be more likely to warn the user.

runn1ng, 2017-10-26:
OP_RETURN can also use empty data, which would cause overestimating.

jhoenicke, 2017-11-06:
Is someone working on this? If not I give it a shot.  It should not be too difficult to get an almost exact estimate (except for the signature length).

prusnak, 2017-11-07:
@jhoenicke Yes, please give it a shot.

prusnak, 2017-11-14:
Fixed via #242

closed 2017-11-14 by prusnak