trezor/trezor-mcu #244: Show path information in GetAddress display dialog

prusnak, 2017-11-14:
If path format is recognized.

prusnak, 2017-11-14:
First attempt implemented in de3b78bd0bb020b953b07a75d3c9d4b75de53f92

Maybe we will be more specific about recognizing custom paths, e.g. `Bitcoin Segwit Account #1 Address 22` is better than `m/49'/0'0'/0/21`

jhoenicke, 2017-11-14:
Slightly related:  I thought about also showing the path when sending to a TREZOR owned address that isn't a valid change address, e.g., when claiming or recovering coins.  Problem is the limited space, though.

saleemrashid, 2017-11-14:
I think we should only show the path if it is not as expected for the input type (e.g. `44'` for P2PKH and `49'` for SegWit) and coin, or the index is abnormally high. If it matches the expected path, we should only show the account.

@jhoenicke ACK, that is especially useful for claiming coins from a Bitcoin fork.

runn1ng, 2017-11-16:
My original "idea" was to try to interpret the path, then show account #, address # and coin info (if found), otherwise warn the user (and possibly show the path). But I realize this is more complicated :)

prusnak, 2017-11-16:
Pushed 3 commits:

cc0896c33451d308a51c2fefe2e8939c8a93fb94 - adds SLIP44 `coin_type` to `CoinInfo`
a713fca85789d23c50ecdd0c13345dd1021023a9 - implements behaviour from @saleemrashid above
3386b16a1c187fb1fa8c48c6328fb6691a8fc570 - detects mismatched coin and SLIP44 `coin_type` in path, shows warning

prusnak, 2017-11-16:
Closing the issue. Feel free to review and comment the mentioned three commits.

closed 2017-11-16 by prusnak