trezor/trezor-mcu #252: Clarify this message: "Encode value of this key?"

fresheneesz, 2017-11-24:
There is a confirmation in here that asks "Encode value of this key?" I got this when trying to use the Trezor Password Manager chrome extension, and I have no idea what it means. Does it mean its encrypting something with the private key? What value is it talking about? What does "of" mean here? This message is pretty worthless at describing what type of action is going to be done here.

saleemrashid, 2017-11-24:
Essentially, it is deriving a key for the string `Activate TREZOR Password Manager?` and using this key to encrypt a constant value `2d650551248d792eabf628f451200d7f51cb63e46aadcbb1038aacb05e8c8aee2d650551248d792eabf628f451200d7f51cb63e46aadcbb1038aacb05e8c8aee`.

The encrypted value is used as a key for decrypting the password database, rather than sending the password database to the TREZOR to decrypt (which would be slow).

fresheneesz, 2017-11-24:
@saleemrashid Thanks for the explanation. Maybe the message should be changed to `"Generate key from value?"` in that case.

prusnak, 2017-11-29:
Not really as it is not being generated but rather computed. Read [SLIP-0011](https://github.com/satoshilabs/slips/blob/master/slip-0011.md) to see how it works.

For now, we'll change the wording to Encode/Decode to Encrypt/Decrypt. If there are better ideas we might reconsider them ...

tsusanka, 2017-11-29:
I've opened a PR changing "encode" to "encrypt", becuase the term "encoding" is wrong. I'm still leaving this issue opened, we should figure out what message to show when the encryption is used as a key derivation function instead of a traditional encryption.

prusnak, 2018-02-20:
Let's close this and reopen if there is a better idea.

closed 2018-02-20 by prusnak