trezor/trezor-mcu #253: Tracking issue for switching to NORCOW

saleemrashid, 2017-11-26:
* [ ] Stop using `nanopb` to generate `Storage`
* [ ] Refactor PIN fail and U2F counter code into separate file
* [ ] Refactor `storage.c` into `storage_` (low-level) and `config_` (high-level) APIs
* [ ] Switch to NORCOW for everything except PIN fails and U2F counter
* [ ] Provide upgrade path for old storage, compatible with all released firmware versions

saleemrashid, 2017-11-26:
I propose that we use the same constants as https://github.com/trezor/trezor-core/blob/master/src/apps/common/storage.py with the following changes:

* We do not use `PIN_FAILS` (and keep this code exactly the same as before)
* We add `HOMESCREEN = const(8)` and `U2F_COUNTER = const(9)`  for the base U2F counter

/cc @jpochyla

saleemrashid, 2017-11-26:
Upon further research, this is problematic because NORCOW relies on having more than one sector.

This means that must do one of the following:

* Use NORCOW for the PIN fails and U2F counter
* Modify NORCOW to work with a single sector

jpochyla, 2017-11-26:
@saleemrashid there's a `pin_fails` branch waiting for review that is more relevant:

https://github.com/trezor/trezor-core/blob/pin_fails/src/apps/common/storage.py
https://github.com/trezor/trezor-core/blob/pin_fails/embed/extmod/modtrezorconfig/storage.c

prusnak, 2017-12-12:
I think this is the better approach for the time being: https://github.com/trezor/trezor-mcu/pull/265 (Jochen revived and finished my older effort)

prusnak, 2017-12-12:
Let's close this for now as the other approach was merged in

closed 2017-12-12 by prusnak