trezor/trezor-mcu #258: Monacoin nowhere to be found in 1.6.0

moleculezz, 2017-12-06:
Hi,

I've upgraded to firmware 1.6.0 because I read the changelog and it said that Monacoin is supported. But I don't see it in the beta wallet. Don't see the other newly introduced coins either except Bitcoin Gold.

Where can I find the Monacoin wallet?

prusnak, 2017-12-06:
Firmware 1.6.0 indeed does support Monacoin. It is not added yet to our web wallet, because we need more testing and we are unsure whether their backend server works with our wallet. Feel free to reach their community whether they at least provide a Monacoin version of Electrum with TREZOR support.

closed 2017-12-06 by prusnak

moleculezz, 2017-12-06:
I was able to find this post http://blog.utyuu.space/2017/11/24/electrum-monatrezor/
But I can't test it out since I don't have Windows and it seems that it doesn't work properly on Mac.