trezor/trezor-mcu #269: Remove non-portable reads from zero page

saleemrashid, 2017-12-14:
The zero page on the Cortex-M3 is aliased to the code, as confirmed by [ARM documentation](http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dai0179b/CHDDBAEC.html), the [datasheet](www.st.com/resource/en/datasheet/stm32f205re.pdf) and my own testing.

Luckily, the first word in the flash is the bootloader stack pointer (`0x20020000`) which results in the first byte being `0x00`. This means that non-portable reads from the zero page have magically worked so far. For example, https://github.com/trezor/trezor-mcu/blob/5812f9865d72c8b10f488f0c7559ddabfa476269/firmware/fsm.c#L332 which results in the call `strlcpy(&storageUpdate.pin, NULL, sizeof(storageUpdate.pin))`. As `strlcpy` does not perform `NULL` checks this should have failed but due to the `0x00` at the start of flash it works as intended.

However, this code is fragile and non-portable and this issue should be fixed.

jhoenicke, 2017-12-16:
Thanks.

Wow, the issue closes itself, if you use the right commit comment :)

closed 2017-12-16 by jhoenicke

saleemrashid, 2017-12-16:
@jhoenicke Have you confirmed that is the only occurrence? In case I was not clear in the issue description, I did not look for other occurences.

Also, yeah, GitHub does cool stuff with PR and commit descriptions :) https://help.github.com/articles/closing-issues-using-keywords/

commit 810443f1 (2017-12-16): Fix NULL pointer access

Fixes #269