trezor/trezor-mcu #280: Specific Error for PinMatrixRequestType_NewSecond not matching PinMatrixRequestType_NewFirst

michal-novacek, 2017-12-20:
When:
- user sets up a new device (resetDevice call)
or
- user changes PIN (changePin call)

The device asks for the new PIN twice for confirmation. There are 3 things which can happen:
1. user enters the same PIN both times => success
2. user enters different pin the second time => error
3. user cancels the action (closes the modal in app) => error

Both 2. and 3. return error "Failure_ActionCancelled". It would be great if case 2. returned a specific error since the action was not cancelled by the user, the user tried to perform the action and it failed because of invalid input.

We can either reuse the existing "Failure_PinInvalid" error type for this case or add a new type.

prusnak, 2017-12-20:
I confirm that we use `Failure_ActionCancelled` at least here:

* https://github.com/trezor/trezor-mcu/blob/f22c849/firmware/reset.c#L63
* https://github.com/trezor/trezor-mcu/blob/f22c849/firmware/fsm.c#L352

Better would be if we created a new type `Failure_PinMismatch` to indicate user has failed to verify the PIN by entering it for the second time.

prusnak, 2018-02-20:
Fixed via fc7189f801f8457cc49bafbacbe546dc3f72d460

closed 2018-02-20 by prusnak