trezor/trezor-mcu #317: Dry-run recovery should display words on screen

slush0, 2018-03-20:
Dry-run recovery should display words on screen to prevent malware lying to device about used words.

jhoenicke, 2018-03-23:
It's probably easier if we always display words on screen for recovery.

Unrelated: dry-run recovery should skip the question "Do you really want to recover the device?" - it is confusing and not necessary.

xbach, 2018-03-24:
Suggestion: dry run recovery should force the advanced/matrix recovery method.

Reasoning:

1. Dry run method is easy and non-destructive to use, making it easy for a user to accidentally compromise their seed. In other words, they might be less careful during dry run than during actual recovery process.

2. Point one is especially exacerbated by the fact that many users accidentally enter their recovery seed in its order, instead of out-of-order as suggested by the screen.

3. This is a wonderful opportunity to show users how advanced recovery works. The dry run process is not as stressful as recovery, therefore users will be less averse to new input methods.