trezor/trezor-mcu #325: [meta] Future improvements of the boot process

prusnak, 2018-03-24:
This is an issue to discuss future improvements of the boot process.

- [ ] update the firmware so it works with the original VTOR table from the bootloader
- [ ] update the firmware so it works in unprivileged mode, most probably enabling the supervisor calls will do the trick: https://github.com/trezor/trezor-mcu/pull/323
- [ ] create a mechanism so SIGNED firmware can indicate to bootloader it needs to use its own VTOR table and/or it needs privileged mode

jhoenicke, 2018-03-24:
The VTOR table has a few gaps, e.g. 0x1c-0x2b.  Can we use one of these as entry points for unprivileged mode?  And/or put the flags there.

Note that flags in the "TRZR"-firmware meta header are not signed, so they cannot be trusted.

Something like if VTOR[0x1c] is 0, the firmware requires privileged mode, otherwise this points to the unprivileged entry point.  Old bootloader will still use the privileged entry at VTOR[0x4] (which contains code that installs mpu, drops privileges and jump to the other entry (or update the bootloader)).

jhoenicke, 2018-03-28:
Random thoughts:

* We need downgrade protection for old firmware or bootloader updates
* Separate bootloader updates?
* Signed bootloader?
* New firmware should always run unprivileged.
* New headers for firmware/bootloader updates, with different magic, download protection info?

saleemrashid, 2018-03-28:
Can we require firmware images to use all the space?

At the moment, you can attach malicious code to the end of the firmware and it is not verified by the bootloader, but the MPU configuration allows you to execute it. (Obviously, this still requires an exploit to jump to the code, but it would be nice to close this security hole).

prusnak, 2018-03-28:
Merged in https://github.com/trezor/trezor-mcu/pull/316 and https://github.com/trezor/trezor-mcu/pull/323