trezor/trezor-mcu #326: ../bootloader/bootloader.bin', needed by `bl_data.h'.

lakshmisatya, 2018-03-26:
Hi All,

Last few days I am struggling to build trezor mcu emulator. After  I followed steps mentioned in readme in Building for development section, My build is getting stopped by showing below message. 
../bootloader/bootloader.bin', needed by `bl_data.h'.  
I installed docker, and tried to compile bootloader also. But nothing works. 
I would like to evaluate trezor emulator with chrome extension. No where I found steps to how to use the trezor emulator for testing. Please help me. 

Regards
Lakshmi

jhoenicke, 2018-03-26:
The emulator build is broken by changes for firmware 1.6.1.  You can go back to the commit just before it: d6f41dba9e41ba49780878dadf157939d2eeb101 to compile the emulator.

Note that you also need to stop the bridge, so that the wallet will use the chrome extension.

lakshmisatya, 2018-03-27:
Hi jhoenicke, 

That works. Thanks for prompt response. 
I also installed trezorctl, when I issue the trezorctl commands, except list option, no option seems to be working. All are showing blank.  "trezorctrl list"  shows udp:127.0.0.1:21324 output. 

Please let me know the steps to how can I use chrome extension with this emulator. I am new to this and clueless. Any help would be highly appreciated. 

Regards
Lakshmikanth

prusnak, 2018-03-27:
Original problem fixed in https://github.com/trezor/trezor-mcu/commit/2f8ea9d5c765f1b8d21a3e2bfe36e2a2283a0bfd

How to configure Chrome Extension to use UDP: https://github.com/trezor/trezor-chrome-extension#udp-connection

closed 2018-03-27 by prusnak