trezor/trezor-mcu #327: Can't load unsigned firmware

Goat42, 2018-03-27:
Hi,

Since last update, I can't load my unsigned firmware anymore. 
I investigated a bit and it seems when the FW is not signed, the new bootloader sets the MPU preventing the bootloader execution. This is performed within bootloader execution...
Do I miss something ?

BR

jhoenicke, 2018-03-27:
Yes, the problem is that the mpu configuration disables execute permission for bootloader and thus even the code that jumps to the firmware is not allowed to execute.  There are more problems, e.g., timer interrupt doesn't work.

I'm currently working on getting unsigned firmware running.  It requires a new bootloader, though (different mpu config and support for timer interrupts).  The bootloader can only be updated by signed firmware, so it will take a while until you can try it.

See #323.

Goat42, 2018-03-27:
So, updating my Trezor I transformed it to a closed platform !!!
Did I miss a warning somewhere?

jhoenicke, 2018-03-28:
Unfortunately yes.  This wasn't intended and it will be fixed.  The firmware/bootloader update was a bit rushed (to fix this security issue with the chips) and the problem with unsigned firmware wasn't seen.

Do you need a quick work-around for this?

prusnak, 2018-03-28:
Fixed by https://github.com/trezor/trezor-mcu/pull/323

closed 2018-03-28 by prusnak