trezor/trezor-mcu #343: SHA256 checksums of downloaded v1.6.1 and self-built firmwares are different

serge-name, 2018-04-28:
Hello,

In #342 I have been told that I need to build the bootloader first. In `v1.6.1` branch build scripts for bootloader and firmware are separate and each of them create separate clean environment in Docker.

So I decided to use `build.sh` from `master`:

```
$ git checkout v1.6.1
HEAD is now at 9588e8f... update bootloader to 1.4.0 via firmware, run unsigned firmware in unprivileged mode
$ git checkout master build.sh
$ ./build.sh bl1.4.0 v1.6.1
[…]
../bootloader/firmware_sign.py -f trezor.bin
Metadata has been added...
Firmware size 430908 bytes
Firmware fingerprint: c2ba53cf5398e142c4d04d8c95538080167510532e4d47c633d052aff22b0f79
Slot #1 is empty
Slot #2 is empty
Slot #3 is empty
make: Leaving directory '/tmp/trezor-mcu-fw/firmware'

Filename    : build/bootloader-bl1.4.0.bin
Fingerprint : be577aec7e8a42d4d8b42f606a76eca043be90699d782dc4126f18d6d4ab11ed
Size        : 32768 bytes (out of 32768 maximum)

Filename    : build/trezor-v1.6.1.bin
Fingerprint : 319c2a804f0da5b3a0e02947fbadc01adf573717281e45b28817fe1adad09310
Size        : 430908 bytes (out of 491520 maximum)
```

Now I am trying to manually check the downloaded (and the built one too) firmware:

```
$ cd ..
$ wget -q https://wallet.trezor.io/data/firmware/1/trezor-1.6.1.bin
$ tail -c +257 trezor-1.6.1.bin | sha256sum
83c3190a94e524ac83a1704eb584a2ab53f8a65a893b1ab52e7135812857c807  -
$ tail -c +257 trezor-mcu/build/trezor-v1.6.1.bin | sha256sum
c2ba53cf5398e142c4d04d8c95538080167510532e4d47c633d052aff22b0f79  -
```

Am I doing something wrong?

[trezor_build_20180428_checksum_mismatch.txt.gz](https://github.com/trezor/trezor-mcu/files/1957876/trezor_build_20180428_checksum_mismatch.txt.gz)

prusnak, 2018-04-28:
Check whether the bootloader is padded with 0x00 and not 0xFF. We
changed that and I am not sure which commit it was.

serge-name, 2018-04-28:
```
$ hexdump -C build/bootloader-bl1.4.0.bin |tail -n 4
00007730  ff ff 24 00 00 24 f4 00  00 24 f4 00 ff ff ff ff  |..$..$...$......|
00007740  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00008000
```

It looks like that it is padded with 0xFF. Am I right?

serge-name, 2018-04-28:
I have found two related commits:
5fa82cc4cb9bc720ed240b1a6ceb63a3c8b97455 — random padding replaced with 0xFF
56ff88a08f6d7ec00157b9b0d778d1b7e9bce3b7 — the padding replaced with 0x00

serge-name, 2018-04-28:
Pavol, thank you for the hint. I have changed the padding to 0x00 and succeded.

For anybody trying to build v1.6.1 firmware themself:

1) use `build.sh` from master
2) apply a patch:

```
--- build.sh.orig   2018-04-27 12:37:37.381926102 -0400
+++ build.sh    2018-04-28 12:53:18.446084637 -0400
@@ -21,6 +21,7 @@
        git submodule update --init --recursive && \
        make -C vendor/libopencm3 && \
        make && \
+       sed -i -r -e 's/xFF/x00/' bootloader/firmware_align.py && \
        make -C bootloader align && \
        cp bootloader/bootloader.bin /$BOOTLOADER_BINFILE && \
        cp bootloader/bootloader.elf /$BOOTLOADER_ELFFILE && \
```

prusnak, 2018-04-28:
Glad you made it. Sorry for the mess! :-/

closed 2018-04-28 by prusnak