trezor/trezor-mcu #346: Building and installing current master results in error

tyrion70, 2018-05-06:
Steps to reproduce after building with ./build.sh on current master:
`git clone https://github.com/trezor/trezor-mcu.git
./build.sh`

`Firmware size 451340 bytes
Firmware fingerprint: a4aa697e02c822593f5f15c1e288a0c7b46d8e8b0fa344c3abe86e27c663db99
...
Filename    : build/bootloader-master.bin
Fingerprint : c8d041b17a061fcbe8e1da3b0c7431069157e5cf3ceb4e2c9d66b4fa816b2dbb
...
Filename    : 
build/trezor-master.bin
Fingerprint : ca2538f8816899444db527daf442209052d316d8547178f063b8840002ab1b6a
Size        : 451340 bytes (out of 491520 maximum)`

Then install
`trezorctl wipe_device
# install default firmware
trezorctl firmware_update  (gives me serial 41BE2521CA... with loader 1.5.0)
trezorctl wipe_device
# install current master firmware 
trezorctl firmware_update -f trezor-mcu/build/trezor-master.bin
..
Firmware fingerprint: a4aa697e02c822593f5f15c1e288a0c7b46d8e8b0fa344c3abe86e27c663db99`

After install and plugging in device it says:
Unofficial firmware detected
when I click " I'll take the risk" it gives above fingerprint and then:

Unknown bootloader detected, Unplug your Trezor and contact our support.

prusnak, 2018-05-06:
You need to add bootloader hash to bl_check.c whitelist manually.

Durendal, 2018-05-07:
I am also experiencing this issue, please advise on the proper course of action to obtain the bootloader hash to insert.

jhoenicke, 2018-05-07:
Bootloader is currently a moving thing, and I'm not sure if my information is still accurate.

But if you have installed a recent self-compiled firmware, and it successfully replaced the bootloader, then unsigned firmware runs in unprivileged mode.  This means it must not try things requiring privileges, in particular setting up timer interrupts or trying to replace bootloader.  Easiest way is to comment out in `trezor.c` function `main` the calls to the functions `check_bootloader`, `timer_init`, `mpu_config`.

Note that if you installed the official firmware 1.6.1, the bootloader-1.4.0 that was installed can't start unsigned firmware at all due to a bug.

tyrion70, 2018-05-07:
@jhoenicke that worked like a charm! Thanks! @Durendal commenting out those functions makes it run for me :) Thanks!

Durendal, 2018-05-07:
this seems to have also resolved the issue for me thankyou very much!

closed 2018-05-09 by prusnak

augorbachev, 2018-11-04:
Hello, I am trying to update the firmware on the official site https://wallet.trezor.io/#/device/web01. after confirming the footprint, I disconnect the trezor and the message "Unknown bootloader detected, Unplug your Trezor and contact our support"
Reread above about add bootloader hash to bl_check.c whitelist manually. where to do it?