trezor/trezor-mcu #349: Screen Flickers On Multiple GetAddresses

MelbourneDeveloper, 2018-05-11:
This code gets 50 addresses. The ShowDisplay flag is set to false.

                    for (uint i = 0; i < 50; i++)
                    {
                        var address = await trezorManager.GetAddressAsync("CRW", 72, false, i, false, AddressType.Bitcoin);
                        await Task.Delay(50);
                    }

This is the message:

`new GetAddress { ShowDisplay = false, AddressNs = path, CoinName = GetCoinType(coinShortcut)?.CoinName, ScriptType = isSegwit ? InputScriptType.Spendp2shwitness : InputScriptType.Spendaddress }`

It does this:

Read: Trezor.Address
Write: Trezor.GetAddress
Delay 50 ms
Read: Trezor.Address
Write: Trezor.GetAddress
Delay 50 ms
Read: Trezor.Address
Write: Trezor.GetAddress
Delay 50 ms
Etc.

However, the Trezor screen goes bananas and the user thinks it's glitching out.

MelbourneDeveloper, 2018-05-11:
My recommendation:

Add a message called "StartWaiting". In this mode the screen will say "Waiting..." and have the animation. The screen will not be updated until the "StopWaiting" message is received, OR some critical message like "SignTx" is received.

prusnak, 2018-05-16:
Fixed in f302b04f311d2a4d0bc2cd3e21eab807857b4a53

closed 2018-05-16 by prusnak

jhoenicke, 2018-05-16:
@prusnak: Have you checked multisig?  IIRC computing multisig addresses sometimes shows the progress dialog, so the layoutHome may be needed for them.

prusnak, 2018-05-17:
@jhoenicke Good catch, reverted the old fix and fixed differently in 233f2f1cfddde95562b731f46dc35bdf7d122127

MelbourneDeveloper, 2018-05-28:
Thanks @prusnak and @jhoenicke 

Is there a way I can monitor when this comment will get rolled in to the next firmware release?