trezor/trezor-mcu #352: Stellar message singing is truncated to 1024 bytes

tsusanka, 2018-05-14:
Stellar implementation currently trims messages to be signed to 1024 bytes.

- Is the truncation needed, since it's done on protobuf level as well? (and is it even possible to send a message larger than 1024?)
- Is forbidding non-ascii chars desired?

Stellar message signing: https://github.com/trezor/trezor-mcu/commit/8e8749dc6826996dd8dbf0fb3332bc817b9de946#diff-7e7c39ac1b5b1f321aec2014d1282466R74

tsusanka, 2018-05-14:
Btw when sending larger message than 1024 I'm getting:

```
RECEIVED <Failure> (19 bytes):
Failure {
    code: 3,
    message: 'string overflow',
}
```

So the check indeed seems redundant

tsusanka, 2018-05-14:
Also, before the verification process, the message should be displayed and confirmed by user

prusnak, 2018-05-14:
Fixed in 29664c4218478a41b063637fc59081534c283a5c

closed 2018-05-14 by prusnak