trezor/trezor-mcu #353: Stellar: protobuf StellarSignMessage.message is inconsistent with StellarVerifyMessage

tsusanka, 2018-05-14:
The `message` field in the `StellarSignMessage` message is declared as UnicodeType, but as BytesType in `StellarVerifyMessage`.

@prusnak Let's make this consistent and declare both as UnicodeType?

prusnak, 2018-05-14:
I would say, let's make it consistent with (Bitcoin) [SignMessage](https://github.com/trezor/trezor-common/blob/master/protob/messages.proto#L529)/VerifyMessage and make the field BytesType

tsusanka, 2018-05-14:
...and done in python-trezor https://github.com/trezor/python-trezor/compare/f97815f69a6a515d0e7ae50554d6f78f1597edaf...master   I hope the `normalize_nfc` usage is correct

prusnak, 2018-05-14:
Fixed in 29664c4218478a41b063637fc59081534c283a5c

closed 2018-05-14 by prusnak