trezor/trezor-mcu #356: Add zcash v3 txs (ZIP 202)

karel-3d, 2018-05-16:
See https://github.com/zcash/zips/blob/master/zip-0202.rst

I am not sure if it is better to be able to create both v1 and v3 transactions, for smoother transition, or just create v3 transactions, for simpler code.

v1 and v2 transactions are valid pre-fork, v3 are valid post-fork.

See also
* https://z.cash/upgrade/overwinter.html
* https://github.com/zcash/zips/issues/151

jhoenicke, 2018-05-16:
> I am not sure if it is better to be able to create both v1 and v3 transactions, for smoother transition, or just create v3 transactions, for simpler code.

We should keep v1 support, also makes it possible to update the firmware before hardfork date or support other zcash-like coins.

I wondered if v3 should also use the signing.c code, but I guess it is the easiest way.   The main change is a different hash function (blake2b), and using bip-143 signatures, but there are also some minor things in the signed data.  Note that v3 has version number 0x80000003, so we could use the high bit as a flag that new kind of signatures should be used.

saleemrashid, 2018-05-16:
@jhoenicke Alternatively, we could have a new `Zcash Overwinter` coin that uses the new transaction version, serialization format and hash function. I have no strong opinion on the matter.

karel-3d, 2018-05-17:
Oh, I did not notice they use bip143 signatures. (https://github.com/zcash/zips/blob/master/zip-0143.rst)

That is good, it means there is no need to add previous txs, great,

> Note that v3 has version number 0x80000003,

Strictly speaking, they call the whole uint32 `header`, the 31 bit `fOverwintered` and version is still just 3, not 0x80000003. So I think even `version` field in `SignTx` should be just 3, not the whole header

jhoenicke, 2018-05-20:
Group ID should be added as a parameter to coins.json.  It should probably be added to CoinInfo (coins.h) but not to CoinType (types.proto).   I separated these structures, so that we don't need to add these fields to our wire protocol.

A different way would be to put group ID into the transaction, but then you need to indicate somehow when it is unexpected.  We do something similar with chain id in ethereum, where the displayed currency symbol depends on chain id.  But as long as we don't have want to support arbitrary group ids, its easier and safer to hard-code the group id in CoinInfo.

If you use version 3 for overwinter, please add a overwintered boolean field to "TransactionType" in  types.proto, or a more general flag field.  This is to avoid complicated checks like "version >= 3 and coin == ZCash".

prusnak, 2018-06-15:
Implemented in https://github.com/trezor/trezor-mcu/commit/f2ad9b921ecc961ebd6805526f092bdf4e165653

closed 2018-06-15 by prusnak