trezor/trezor-mcu #358: Stellar: message signing allows an arbitrary message

tsusanka, 2018-05-22:
The sign/verify functions allow signing arbitrary message. This is dangerous. The reasons are discussed at https://github.com/trezor/trezor-core/pull/90#issuecomment-390790239 but I'll copy it here:

1) separation of message signing and transaction signing. When I'm signing arbitrary Bitcoin message, I can be sure that it won't lead to coinloss. Stellar implementation signs whatever you give to it, even a transaction digest.

2) separation of coins on the same curve. For example Litecoin has a different message prefix. Stellar doesn't do that, it allows signing any message with any key derived from seed on ed25519 curve.

Usually this is solved by prefixing the message with a header: https://github.com/trezor/trezor-mcu/blob/8bdf338f32a881d5712a138e6c2d667c035264cc/firmware/crypto.c#L117 or in trezor core: https://github.com/trezor/trezor-core/blob/master/src/apps/common/signverify.py#L12 

@zulucrypto we will remove this at the moment. But, do you know if Stellar is dealing with this issue somehow (maybe the documentation is addressing this issue)? If not, do you think you could create an issue on their side?

prusnak, 2018-05-22:
> Separation of coins on the same curve

For example Lisk uses the same curve. So you can use Stellar SignMessage feature to sign a legit Lisk transaction using the same private key. That's why adding a prefix is important.

Let's remove the StellarSignMessage/StellarVerifyMessage for now. We can readd this later once there is a need for this and there is a Stellar standard specifying the prefix.

zulucrypto, 2018-05-22:
> Let's remove the StellarSignMessage/StellarVerifyMessage for now. We can readd this later once there is a need for this and there is a Stellar standard specifying the prefix.

Sounds good, I'll work through them to propose a standard that uses `Stellar Signed Message:\n` to match the existing ones.

prusnak, 2018-05-22:
Removed Stellar{SignMessage,MessageSignature,VerifyMessage} in
https://github.com/trezor/trezor-mcu/commit/ae82735b6b7ecbe46a0baf3243ccd0f3e73ca21d

closed 2018-05-22 by prusnak