trezor/trezor-mcu #359: Windows Device Drivers Stacking Up

MelbourneDeveloper, 2018-06-03:
After the last firmware/bridge update, my UWP (Windows 10) app stopped being able to connect to the Trezor while my Android app has no issues. I'm still trying to diagnose the problem, but it may be related to this. I currently have two Hid devices installed with the same VID and PID for Trezor. Windows is complaining that there is an ambiguous match. The ID of the device, according to windows changed after the last firmware/bridge upgrade.

[image]

[image]

prusnak, 2018-06-03:
I don't think this is a trezor issue. I'd suggest contacting Microsoft support.

MelbourneDeveloper, 2018-06-07:
@prusnak you are probably 100% correct. Nevertheless, this problem broke my app. 

Windows is returning 3 Hid Ids for my Trezor:
\\?\USB#VID_534C&PID_0001#2EE5252107F7BF65437A78B1#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
\\?\HID#VID_534C&PID_0001&MI_01#7&1cc30b55&1&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
\\?\HID#VID_534C&PID_0001&MI_00#7&2ff32793&1&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}

When I narrowed down the correct Id which is \\?\HID#VID_534C&PID_0001&MI_00#7&2ff32793&1&0000#{4d1e55b2-f16f-11cf-88cb-001111000030} the app started working again.

From an app point of view, how can I tell which one is the correct Trezor? The Id changed with the last firmware update. So, this problem is bound to happen again on the next firmware update. I'm currently selecting on Vid and Pid, but this only narrows the choice down to 3 devices, so I need a 3rd part.

Will selecting on "MI_00", be enough to stop the problem from happening on the next firmware update?

MelbourneDeveloper, 2018-06-07:
According to the shitty Windows 10 API, all these devices **_are_** enabled, but none of them are the **_default_**. So, I'm left taking a wild guess.

MelbourneDeveloper, 2018-06-07:
@prusnak , actually, the problem is nowhere near as severe as I thought...

I iterated through the three Ids, attempted to connect to each, and then continued on with the one that works. It's allowed me to fix my app and this should be future proof for the next firmware update. Just thought I'd log this in case other experience the same issue.

prusnak, 2018-06-07:
Yes, select `MI_00` which is guaranteed to be always the right interface.

closed 2018-06-07 by prusnak