trezor/trezor-mcu #360: Canceling "show address" from PC will toggle QR code instead of canceling

matejcik, 2018-06-04:
it seems that T1 interprets `Cancel` message as "button 2".

That means that when showing address, sending Cancel from PC side will switch between QR code and address text, instead of canceling action.

Test case here: https://github.com/trezor/python-trezor/commit/ff5402842ca5650f70811f4dcb86ce1282fcea34#diff-680dbc8e79e141d1df4f7d7124c28446

karel-3d, 2018-06-05:
Wasn't this already fixed somewhere? I remember this vaguely

karel-3d, 2018-06-05:
Oh I am confusing it with

https://github.com/trezor/trezor-mcu/pull/287

matejcik, 2018-06-05:
that looks like the same issue, yes
however this test fails on current master. I can push it to python-trezor master and make trezor-mcu fail, if you like ;)

prusnak, 2018-06-05:
Yes, please add the test to python-trezor master.

saleemrashid, 2018-06-05:
Solution should be as simple as adding a `protectAbortedByCancel`, right?

prusnak, 2018-06-05:
Fixed as per Saleem's suggestion in df0bad4f09bff651023562e22735ab937a576149

closed 2018-06-05 by prusnak