trezor/trezor-mcu #361: StellarGetPublicKey message returns 'Failure': 'Action cancelled by user'

szymonlesisz, 2018-06-07:
I'm trying to use "StellarGetPublicKey" message inside trezor-connect with Trezor 1 and Bridge-v2
After i send message to device: "StellarGetPublicKey", { address_n: ... } 
device will show address on display but response from device is "Failure": "Action cancelled by user".
I think i should get "ButtonRequest" instead.

When i refresh communication, after send "Initialize" message i get another "Failure": "Action cancelled by user" response

szymonlesisz, 2018-06-11:
Update:

This happens only when i send values which are not hardended (example: [1, 0, 0])
With hardended values it works properly. 

So the issue here is invalid error message. 
Should be "Failed to derive private key" (like in NEMGetAddress)

prusnak, 2018-06-11:
Fixed in https://github.com/trezor/trezor-mcu/commit/217d5dd01d4b85003ac163fdeae969e07431a237

closed 2018-06-11 by prusnak