trezor/trezor-mcu #362: Trezor Server

MelbourneDeveloper, 2018-06-11:
This an idea more than a feature request. The Trezor Bridge allows communication from JavaScript websites to the Trezor without the need for low level USB transfer. I'm not sure how this works, but what would be great is if the Bridge were extended to serve up the data over https. I.e. a web server would be installed on the host machine that would talk to the Trezor, and any apps could talk to the REST service with the same messages in JSON as the protobuf definitions. 

The reason I think this is necessary is that iOS devices cannot directly communicate with the Trezor. So, if someone installed the server on the Mac, they could connect the Trezor to their Mac but use the Trezor from their iPhone/iPad.

iPhones and iPads are a very large slice of the market. A lot of people who have Trezors don't have an Android phone which is capable of talking to the Trezor so these people are not able to use Trezor with phone apps. 

Here is some sample code from my Trezor library:

`var readBuffer = await _TrezorHidDevice.ReadAsync();`

_TrezorHidDevice is an interface. So, the class that talks to the hardware wallet could easily be replaced with a class that talks to the REST service instead. The same can be said with this code:

`await _TrezorHidDevice.WriteAsync(range);`

In the long run, I'm going to need something like this so I may have to end up writing it myself. But, it seems like something that would be better for you guys to do because I'd really struggle with getting users to install the https server on their machine.

prusnak, 2018-06-11:
This is really cumbersome. Feel free to create your own fork of the bridge which servers the data over HTTPS, but I don't think that is a good idea as it is very user unfriendly.

closed 2018-06-11 by prusnak

saleemrashid, 2018-06-12:
> So, if someone installed the server on the Mac, they could connect the Trezor to their Mac but use the Trezor from their iPhone/iPad.

Given that the TREZOR requires physical presence anyway, why wouldn't you simply use the Mac?

MelbourneDeveloper, 2018-06-29:
@saleemrashid , the quick answer is yes.  The long answer is no. 

I will be building a version of my app for Mac first. But, there's still a tonne of people out there that use iPhones and they want apps for their phone. My app is a phone first app and designed for phones. I'd really like to get it going on iPhone but, it's currently not possible because iPhones won't connect to a USB device. 

I carry my app around on my phone all the time. I don't turn my computer on to use it. Actually, my app only connects to the Trezor at first to grab the addresses, and then you can disconnect the OTG cable. So, if someone were using an iPhone with a Trezor connected to their Mac, they could get all the addresses from their Mac and then continue to use the iPhone version from then on. They wouldn't need to turn on their computer to get balances.

@prusnak , it's not really something that the users would experience _per se_. The server would act as a REST server, and apps like Hardfolio would connect to it as a client. The user really wouldn't have to do anything.

MelbourneDeveloper, 2018-08-01:
@prusnak , I would still need to connect to the device to get the XPUB. But also, I'm attempting to not store the XPUB in storage for the point of extra privacy.