trezor/trezor-mcu #365: Firmware too large

saleemrashid, 2018-06-18:
As of b73e18e5732e3493edbd0860577f39053c86826a (`trezor-crypto` updated in f2ad9b921ecc961ebd6805526f092bdf4e165653), the firmware is too large.

jhoenicke, 2018-06-18:
Using `-Os` seems to be enough for now.

Other ideas to reduce size:
- disable blake & groestl in hashers.c (they are not used yet, right?)
- don't use precomputed tables for nist/secp (38 kB).
- separate bootloader update (32kB).
- increase  allowed firmware size (512kB, but requires separated bootloader update).

saleemrashid, 2018-06-18:
* Decred (BLAKE) and Groestlcoin (Grøstl) are in firmware `master` so we probably shouldn't disable those hashers
* Disabling precomputed tables for NIST P-256 seems like a good trade-off (won't affect the majority of users)
* Compressing the bootloader might also be a possibility (e.g. including a "diff" of the bootloader to the firmware)

prusnak, 2018-06-19:
1) ack
2) we cannot disable precomputed for NIST and keep it enabled for SECP currently (because of how the ecdsa code is currently written)
3) hm, interesting idea

prusnak, 2018-06-19:
For now fixed in a46de7b5916b74cc31a0aed1396ca47e4e4fed50

We can afford using the full 1M of Flash after the next bootloader update.

closed 2018-06-19 by prusnak