trezor/trezor-mcu #367: Stellar: modify account fields to use addresses

tsusanka, 2018-06-19:
Already merged in trezor-common: https://github.com/trezor/trezor-common/issues/135#issuecomment-397289993

To be merged in trezor-core: https://github.com/trezor/trezor-core/commit/6f0d18e1fe8aa4e2d269eeb55aa087c8d294102b

Tests are in python-trezor in the [tsusanka/stellar-accounts-addresses](https://github.com/trezor/python-trezor/tree/tsusanka/stellar-accounts-addresses) branch

prusnak, 2018-06-20:
python-trezor merged in https://github.com/trezor/python-trezor/commit/10bbb57c865ed514a01daa7b4e821207b62c7f89

prusnak, 2018-06-20:
This would need much more refactoring than I originally thought :-/

I pushed the trezor-common updated submodule in 2ec6129da5eb1eaa2adae9cc1e283140d1ee19f5 effectively breaking the disabled Stellar code, but we need to refactor that anyway :-/

zulucrypto, 2018-06-22:
Has anyone started on refactoring this in trezor-mcu? If not, I'll take a stab at it this weekend.

prusnak, 2018-06-22:
Please do so! Thank you!

PR #372 (2018-06-27): Stellar: use strings instead of bytes for addresses
 * Use the new protobuf messages that send accounts as strings (fixes #367)
 * `stellar_signingAbort()` now takes an error message
 * Operations now check if a transaction is being signed (fixes #368)
 * Operations now return false on error or if the user cancels
 * Stellar `fsm_*` methods now check operation result and return early if the operation fails

This PR also re-enables the stellar code in the firmware.
commit d1a48f32 (2018-06-27): Stellar: use strings instead of bytes for addresses (#372)

 * Use the new protobuf messages that send accounts as strings (fixes #367)
 * `stellar_signingAbort()` now takes an error message
 * Operations now check if a transaction is being signed (fixes #368)
 * Operations now return false on error or if the user cancels
 * Stellar `fsm_*` methods now check operation result and return early if the operation fails

This PR also re-enables the stellar code in the firmware.
closed 2018-06-27 by prusnak