trezor/trezor-mcu #368: Stellar operations do not check if signing process has started

saleemrashid, 2018-06-19:
The various `fsm_msgStellar*Op` functions do not check `stellar_signing`. As such, they can be called to perform operations on the private keys, even when the PIN has not been entered.

cc @zulucrypto

prusnak, 2018-06-19:
JFYI I disabled Stellar in the current master: https://github.com/trezor/trezor-mcu/commit/16ad5ce85401de53864b61afca88e1efebb7219a (it should be pretty easy to re-enable it by doing the reverse operation of the commit).

We'll try to stabilize the code and include it in the next firmware update.

zulucrypto, 2018-06-20:
@prusnak - Sounds good, would you like me to work on a PR that adds this check or is someone else already on it?

prusnak, 2018-06-20:
Feel free to do so. Thanks!

prusnak, 2018-06-20:
I broke the current Stellar code - we'd need to refactor it to use addresses instead of the pubkeys - see https://github.com/trezor/trezor-mcu/issues/367 for more info.

PR #372 (2018-06-27): Stellar: use strings instead of bytes for addresses
 * Use the new protobuf messages that send accounts as strings (fixes #367)
 * `stellar_signingAbort()` now takes an error message
 * Operations now check if a transaction is being signed (fixes #368)
 * Operations now return false on error or if the user cancels
 * Stellar `fsm_*` methods now check operation result and return early if the operation fails

This PR also re-enables the stellar code in the firmware.
commit d1a48f32 (2018-06-27): Stellar: use strings instead of bytes for addresses (#372)

 * Use the new protobuf messages that send accounts as strings (fixes #367)
 * `stellar_signingAbort()` now takes an error message
 * Operations now check if a transaction is being signed (fixes #368)
 * Operations now return false on error or if the user cancels
 * Stellar `fsm_*` methods now check operation result and return early if the operation fails

This PR also re-enables the stellar code in the firmware.
closed 2018-06-27 by prusnak