trezor/trezor-mcu #380: Signed personal Ethereum messages starting with "0x" can not be verified

sterlu, 2018-07-03:
When signing an Ethereum message starting with "0x" the message can not be verified. Other messages (ie. "test") are signed and verified correctly. 

I've used https://www.myetherwallet.com/signmsg.html to sign & verify, as well as Trezor Connect v4 & 0x.js isValidSignature. I am using Trezor One with firmware version 1.6.2

### Steps to reproduce: 
- sign a message starting with "0x" such as "0x12345" on https://www.myetherwallet.com/signmsg.html
- try to verify returned signature on same site

Small disclamer: Trying to sign and verify "test" on MEW only works after deleting `"version": 3` from returned signature.

prusnak, 2018-07-03:
I don't think this is a Trezor issue, but rather a MyEtherWallet issue.

sterlu, 2018-07-03:
As mentioned, I've tried alternate ways to verify the signature. It is verified correctly if the message is a simple string, but fails when the message starts with "0x". 

Other verification methods I've tried (such as the mentioned 0x.js isValidSignature) verify signatures made with other wallets (MetaMask, Ledger, Keystore) correctly which led me to believe this might be an issue on Trezor's side.

sterlu, 2018-07-13:
Hi again, I eventually figured it out by converting hex messages into buffers. I created a script testing a few message formats. The script signs messages on Trezor, then runs ecrecover on received signatures and compares the address of resulting public key with ledger address. Here are the results:

```
import ethUtil from 'ethereumjs-util';
// trezor login, etc.                      
await testMessageSigning('test');                            // OK
await testMessageSigning('12345');                           // OK
await testMessageSigning('0x12345');                         // FAIL
const dataBuff = ethUtil.toBuffer('0x12345');
await testMessageSigning(dataBuff)                           // OK
const msgHashBuff = ethUtil.hashPersonalMessage(dataBuff);
await testMessageSigning(msgHashBuff)                        // OK (ecrecover ok, but is signing a different message bascally so if you're not the one doing the verification this will not solve your problem)
await testMessageSigning(ethUtil.bufferToHex(msgHashBuff));  // FAIL (again failing with strings starting with '0x'
```

I am leaving the issue open as I am not sure whether this is the expected behavior.

prusnak, 2018-07-16:
So this is a Javascript error after all and not a TREZOR issue.

Please open an issue for https://github.com/trezor/connect and post the output from above and also the definition of your `testMessageSigning` function.

closed 2018-07-16 by prusnak