trezor/trezor-mcu #384: bootloader: jump_to_firmware can lead to undefined behavior

mcudev, 2018-07-17:
currently the `jump_to_firmware` function can lead to undefined behavior if an exception is taken after `SCB_VTOR` is set and before the firmware's `reset_handler` has completed. what i mean is that the firmware exception handlers are written in C and the sections that get initialized in SRAM are put there by the `reset_handler`. this is an informational issue right now because i don't see this being exploitable with the firmware's current set of exception handlers. the model t already considered this possibility and handles it differently.

to see this in action in a dev board, set a break point at https://github.com/trezor/trezor-mcu/blob/master/util.h#L65 and touch the reset pin on your board to cause clock instability which will trigger the `nmi_handler` for the clock security system. since the firmware's vectors are now in use, when you step in the debugger, you'll be transported to the exception handler at the firmware's flash memory address. thus, we're running firmware code before completing the bootloader and the subsequent firmware initialization.

jhoenicke, 2018-07-18:
What do you suggest? Moving the SCB_VTOR initialization into firmware would make it incompatible with old firmwares (but we should think about this once we actively break support for old firmware).

Not writing the exception handlers in C is probably not what we want.

Also if there is a clock instability in the bootloader before display is initialized, the handler would also not work.

mcudev, 2018-07-18:
I think something along the lines of what we did with the Model T where interrupts are disabled, registers cleared, setup the stack, and then repoint the VTOR. See also https://github.com/trezor/trezor-core/issues/281. The problem that I am pointing out is that right now interrupts are enabled and can lead to this unexpected change in control flow.

I mentioned the C code just because in the worst case we're running them before their environment has been initialized and thus _could_ lead to bad behavior. All the handlers that I see in the official firmware are pretty much do nothing handlers right now. I don't see much harm _right now_, but figured I'd put the info out to prevent a future foot-gun. SysTick uses a .bss type variable but that's ok because VTOR is reset after memset_reg anyways. It would just be more of a problem if some future change in the handlers made them depend on data initialized by the reset handler. For example, if USB was switched from polling to interrupt driven, then this could maybe become more useful to an attacker (depending, of course, on the actual changes).

The clock instability was just one easy way that I knew of to easily trigger the demonstration. Touching the pin with your finger actually does cause instability and works to prove my point. The NMI handler runs because the clock gets automatically switched back over to the HSI. It's just that the code in the NMI handler is safe right now because it only uses pointers to strings in flash memory. As an aside, maybe there's another way to generate a similar circumstance via the USB data lines or some other external interrupt. It would just take more work to try to make that happen because it would have to be done via some custom circuit to drive the USB data lines since at that point in the code the device hasn't registered with the OS yet to be able to try something simple with like libusb. The USB exception handlers are do nothing handlers for Trezor One because it's handled with polling, so it wouldn't be exploitable right now.