trezor/trezor-mcu #395: docker build issue

juttTechnology, 2018-08-06:
here is log from the last

make: Entering directory '/src/build/bootloader/vendor/libopencm3'
  GENHDR  include/libopencm3/lpc17xx/irq.json
  GENHDR  include/libopencm3/stm32/f2/irq.json
  GENHDR  include/libopencm3/stm32/f3/irq.json
  GENHDR  include/libopencm3/stm32/l4/irq.json
  GENHDR  include/libopencm3/stm32/f4/irq.json
  GENHDR  include/libopencm3/stm32/f0/irq.json
  GENHDR  include/libopencm3/stm32/f7/irq.json
  GENHDR  include/libopencm3/stm32/l0/irq.json
  GENHDR  include/libopencm3/stm32/l1/irq.json
  GENHDR  include/libopencm3/stm32/f1/irq.json
  GENHDR  include/libopencm3/efm32/tg/irq.json
  GENHDR  include/libopencm3/efm32/gg/irq.json
  GENHDR  include/libopencm3/efm32/g/irq.json
  GENHDR  include/libopencm3/efm32/lg/irq.json
  GENHDR  include/libopencm3/sam/3u/irq.json
  GENHDR  include/libopencm3/sam/3s/irq.json
  GENHDR  include/libopencm3/sam/3a/irq.json
  GENHDR  include/libopencm3/sam/3n/irq.json
  GENHDR  include/libopencm3/sam/3x/irq.json
  GENHDR  include/libopencm3/sam/d/irq.json
  GENHDR  include/libopencm3/vf6xx/irq.json
  GENHDR  include/libopencm3/lm3s/irq.json
  GENHDR  include/libopencm3/lpc13xx/irq.json
  GENHDR  include/libopencm3/lpc43xx/m0/irq.json
  GENHDR  include/libopencm3/lpc43xx/m4/irq.json
  BUILD   lib/stm32/f2
  CP      libopencm3_stm32f2.ld
  CC      gpio.c
  CC      rcc.c
  CC      desig.c
  CC      crc_common_all.c
  CC      dac_common_all.c
  CC      dma_common_f24.c
  CC      gpio_common_all.c
  CC      gpio_common_f0234.c
  CC      i2c_common_v1.c
  CC      iwdg_common_all.c
  CC      rtc_common_l1f024.c
  CC      spi_common_all.c
  CC      spi_common_l1f124.c
  CC      timer_common_all.c
  CC      timer_common_f0234.c
  CC      timer_common_f24.c
  CC      usart_common_all.c
  CC      usart_common_f124.c
  CC      flash_common_f234.c
  CC      flash_common_f24.c
  CC      hash_common_f24.c
  CC      crypto_common_f24.c
  CC      exti_common_all.c
  CC      rcc_common_all.c
  CC      rng_common_v1.c
  CC      usb.c
  CC      usb_standard.c
  CC      usb_control.c
  CC      usb_fx07_common.c
  CC      usb_f107.c
  CC      usb_f207.c
  CC      usb_msc.c
  CC      vector.c
  CC      systick.c
  CC      scb.c
  CC      nvic.c
  CC      assert.c
  CC      sync.c
  CC      dwt.c
  AR      libopencm3_stm32f2.a
make: Leaving directory '/src/build/bootloader/vendor/libopencm3'
  DEP     gen/fonts.d
  DEP     gen/bitmaps.d
  DEP     timer.d
  DEP     supervise.d
  DEP     memory.d
  DEP     util.d
  DEP     setup.d
  DEP     serialno.d
  DEP     rng.d
  DEP     oled.d
  DEP     layout.d
  DEP     buttons.d
  AS      startup.o
  CC      buttons.o
  CC      layout.o
  CC      oled.o
  CC      rng.o
  CC      serialno.o
  CC      setup.o
  CC      util.o
  CC      memory.o
  CC      supervise.o
  CC      timer.o
  CC      gen/bitmaps.o
  CC      gen/fonts.o
  AR      libtrezor.a
make: Entering directory '/src/build/bootloader/bootloader'
  DEP     ../vendor/trezor-crypto/memzero.small.d
  DEP     ../vendor/trezor-crypto/sha2.small.d
  DEP     ../vendor/trezor-crypto/secp256k1.small.d
  DEP     ../vendor/trezor-crypto/ecdsa.small.d
  DEP     ../vendor/trezor-crypto/bignum.small.d
  DEP     usb.d
  DEP     signatures.d
  DEP     bootloader.d
  CC      bootloader.o
  CC      signatures.o
  CC      usb.o
  CC      ../vendor/trezor-crypto/bignum.small.o
  CC      ../vendor/trezor-crypto/ecdsa.small.o
  CC      ../vendor/trezor-crypto/secp256k1.small.o
  CC      ../vendor/trezor-crypto/sha2.small.o
  CC      ../vendor/trezor-crypto/memzero.small.o
  LD      bootloader.elf
  OBJCOPY bootloader.bin
./firmware_align.py bootloader.bin
make: Leaving directory '/src/build/bootloader/bootloader'
make: Entering directory '/src/build/bootloader/vendor/nanopb/generator/proto'
protoc --python_out=. nanopb.proto
/usr/local/bin/protoc: 4: /usr/local/bin/protoc: Syntax error: "(" unexpected
Makefile:4: recipe for target 'nanopb_pb2.py' failed
make: *** [nanopb_pb2.py] Error 2
make: Leaving directory '/src/build/bootloader/vendor/nanopb/generator/proto'

prusnak, 2018-08-06:
`./build.sh` works for me with current master.

Which command do you run?

juttTechnology, 2018-08-06:
MEMORY_PROTECT=0 ./build.sh

now i am installing new os. so i will give you feedback tomorrow

juttTechnology, 2018-08-07:
i installed the ubuntu mate on raspbery pi 3 after that i got the same output with error as i mentioned above.

juttTechnology, 2018-08-08:
now i install the ubuntu 18.04.1 in virtualbox and everzthing is compilied successfuly using this method
https://mcudev.github.io/trezor-dev-kit/
after that i just flash the combined.bin using stm32 st link utility.

now the problem is i got screen of unofficial firmware detected. how i solve this issue .

closed 2018-08-09 by juttTechnology