trezor/trezor-mcu #397: Self compiled firmware issue unofficial  firmware detected

juttTechnology, 2018-08-09:
self complied firmware give me the error of unofficial firmware detected.
i followed this method 
https://mcudev.github.io/trezor-dev-kit/

after that i got unofficial firmware issue then i got this 
https://github.com/trezor/trezor-mcu/issues/346#issuecomment-387036589

after that i complie again after comment this three line but still have same issue .
i am using stm32 link utility software. i am flashing combined.bin.

any help will be appreciated.

thank you

jhoenicke, 2018-08-09:
The "Unofficial firmware" warning and the hash confirmation is expected.
If you get "Unknown bootloader detected" you need to white-list your own bootloader (`known_bootloader` in firmware/bl_check.c), or just skip the check.  Since you should compile with MEMORY_PROTECT=0 for your own hardware, this should already be the case.

Once you compile without  MEMORY_PROTECT=0 and install that, your device is locked and stm32 link doesn't work any more. You have to hope that the bootloader is working and use trezorctl to flash a new firmware.

juttTechnology, 2018-08-09:
The "Unofficial firmware" warning and the hash confirmation is expected.
this is the thing i like to know how i can do that.
yes this is my own hardware. and i always compile with MEMORY_PROTECT=0. so my bootloader is not locked.

jhoenicke, 2018-08-09:
Create your own signing keys with `bootloader/firmware_sign.py -g`.  

Add your own keys to `signatures.c` and `firmware_sign.py` (you need at least three keys of your own).

Sign the firmware with `bootloader/firmware_sign.py -s` three times with different keys.

jhoenicke, 2018-08-09:
Or alternatively change signatures_ok in signatures.c  to always return SIG_OK.

juttTechnology, 2018-08-09:
i used the first method as you described, now i will mention all the steps what i did, but still same issue that unofficial firmware detected 

git clone https://github.com/trezor/trezor-mcu.git
cd trezor-mcu/

// here i use this command three time to generate the keys (public,Private) 
bootloader/firmware_sign.py -g
// then i add Public keys into the signature.c and firmware?sign.py according to there strucutre 

git submodule update --init --recursive
make -C vendor/libopencm3/ lib/stm32/f2
export MEMORY_PROTECT=0
make
make -C bootloader/ align
make -C vendor/nanopb/generator/proto/
make -C firmware/protob/
make -C firmware/ sign

// at this point i used this command 

bootloader/firmware_sign.py -f  ./firmware/trezor.bin -s

then i got option to select solt number from 1-3. i choosed 1 then i got option to insert Private key.
after inserting the Key i saw message Valid something and so.
i did this three times for three slot and for 3 private keys .then 

cp bootloader/bootloader.bin bootloader/combine/bl.bin
cp firmware/trezor.bin bootloader/combine/fw.bin
cd bootloader/combine/ && ./prepare.py

After that i just flash the combine.bin file.

so this was the procedure but the result is same no official firmware detected.

juttTechnology, 2018-08-09:
one thing i like to mention that in signature.c  had 5 public keys and i just update the last 3 but first two is still same. in firmware_sign.py had only 3 so i replaced all three of them..

signature.c still have first 2 old keys so may be this is the issue .

jhoenicke, 2018-08-09:
The public keys in signature.c and firmware_sign.py must be in the same order and must be the same keys.    So the key marked with 1: in firmware_sign.py must be the first entry in signature.c.  The reason is that the keys are identified by their position in the array.   Also PUBKEYS must be equal to the number of public keys (but SLOTS in firmware_sign.py must always be 3, as it is hardcoded in signature.c).

closed 2018-08-10 by prusnak

juttTechnology, 2018-08-10:
perfect..everything is working now..thank you