trezor/trezor-mcu #401: Implement support for EIP-712

rmeissner, 2018-08-15:
Currently there is no consensus on how to properly sign messages. EIP-712 proposes a new standard.

EIP-712: https://github.com/ethereum/EIPs/blob/master/EIPS/eip-712.md

Issue for T2: https://github.com/trezor/trezor-core/issues/122

Reference: https://github.com/trezor/trezor-mcu/issues/312

prusnak, 2018-08-15:
If you plan to implement this, I suggest you start with Model T first, then proceed by creating an unit test in python-trezor similar to https://github.com/trezor/python-trezor/blob/master/trezorlib/tests/device_tests/test_msg_ethereum_signmessage.py

marc0olo, 2018-08-17:
@prusnak could you elaborate the necessary steps to implement this feature in a bit more detail? if there is nobody that tackles this I might give it a try

but I neither developed for trezor nor have much experience with python. so I need a few hints what needs to be done

prusnak, 2019-04-15:
We switch to the monorepo, please send PR there: https://github.com/trezor/trezor-firmware

closed 2019-04-15 by prusnak