trezor/trezor-mcu #405: MacOS emulator support

backslash47, 2018-08-30:
Hello, 

is there a way how to run the emulator under MacOS?

prusnak, 2018-08-30:
It might work. Try running `EMULATOR=1 ./script/fullbuild`, but this was never tested before.

backslash47, 2018-08-30:
Thanks. I was able to build the emulator and run it under MacOS with some small modifications to the build script.

closed 2018-08-30 by backslash47

prusnak, 2018-08-30:
Please share the modifications here, so others can see it too, when they hit the same issues.

backslash47, 2018-08-30:
No problem.

1. Both sdl2 and sdl2_image needs to be installed using Homebrew.
2. Apply this patch: https://gist.github.com/backslash47/fee8f1229f68aa6dd1f952d0cc70b4a9
3. Build with EMULATOR=1 TREZOR_TRANSPORT_V1=1 DEBUG_LINK=1 ./script/cibuild
    - it will fail because trezor.bin is not there, but the emulator is called trezor.elf
4. run with ./firmware/trezor.elf

backslash47, 2018-08-30:
2. OK, that's working
3. fullbuild requires commit hashes as parameters, what should I give them?

prusnak, 2018-08-30:
`./script/fullbuild HEAD HEAD`

backslash47, 2018-08-30:
Ok, the final required steps are:

1. `brew install sdl2 sdl2_image coreutils`
2. apply patch https://gist.github.com/backslash47/fee8f1229f68aa6dd1f952d0cc70b4a9
3. `EMULATOR=1 TREZOR_TRANSPORT_V1=1 DEBUG_LINK=1 ./script/fullbuild HEAD HEAD`