trezor/trezor-mcu #417: when run: ./build-emulator.sh, get error ImportError: No module named 'click'

seelezlk, 2018-10-09:
~/trezor-mcu$ sudo ./build-emulator.sh
Sending build context to Docker daemon  146.6MB
Step 1/7 : FROM debian:9
 ---> f2aae6ff5d89
Step 2/7 : RUN apt-get update &&     apt-get install -y     build-essential curl unzip git python3 python3-pip     libsdl2-dev libsdl2-image-dev
 ---> Using cache
 ---> eef258c0c56b
Step 3/7 : ENV PROTOBUF_VERSION=3.4.0
 ---> Using cache
 ---> 20fa531fb6b7
Step 4/7 : RUN curl -LO "https://github.com/google/protobuf/releases/download/v${PROTOBUF_VERSION}/protoc-${PROTOBUF_VERSION}-linux-x86_64.zip"
 ---> Using cache
 ---> cf126254ad21
Step 5/7 : RUN unzip "protoc-${PROTOBUF_VERSION}-linux-x86_64.zip" -d /usr
 ---> Using cache
 ---> e7be8811ae59
Step 6/7 : RUN pip3 install "protobuf==${PROTOBUF_VERSION}" ecdsa
 ---> Using cache
 ---> 9570c8e7e7df
Step 7/7 : RUN ln -s python3 /usr/bin/python
 ---> Using cache
 ---> 444dd5da8d88
Successfully built 444dd5da8d88
Successfully tagged trezor-mcu-build-emulator64:latest
Cloning into 'trezor-mcu'...
remote: Enumerating objects: 30, done.
remote: Counting objects: 100% (30/30), done.
remote: Compressing objects: 100% (21/21), done.
remote: Total 6098 (delta 13), reused 19 (delta 9), pack-reused 6068
Receiving objects: 100% (6098/6098), 2.37 MiB | 3.08 MiB/s, done.
Resolving deltas: 100% (4537/4537), done.
Already on 'master'
Your branch is up-to-date with 'origin/master'.
Submodule 'libopencm3' (https://github.com/libopencm3/libopencm3.git) registered for path 'vendor/libopencm3'
Submodule 'vendor/nanopb' (https://github.com/nanopb/nanopb.git) registered for path 'vendor/nanopb'
Submodule 'trezor-common' (https://github.com/trezor/trezor-common.git) registered for path 'vendor/trezor-common'
Submodule 'trezor-crypto' (https://github.com/trezor/trezor-crypto.git) registered for path 'vendor/trezor-crypto'
Submodule 'trezor-qrenc' (https://github.com/trezor/trezor-qrenc.git) registered for path 'vendor/trezor-qrenc'
Cloning into '/trezor-mcu/vendor/libopencm3'...
Cloning into '/trezor-mcu/vendor/nanopb'...
Cloning into '/trezor-mcu/vendor/trezor-common'...
Cloning into '/trezor-mcu/vendor/trezor-crypto'...
Cloning into '/trezor-mcu/vendor/trezor-qrenc'...
Submodule path 'vendor/libopencm3': checked out 'b0e050d10d12c42be031c34822117cfd3c5a0ea7'
Submodule path 'vendor/nanopb': checked out 'cdb61612197feca5391593d051c8d4892e940228'
Submodule path 'vendor/trezor-common': checked out '41e4a84b5b01d03e980f84fab29c8f0b0ec948f5'
Submodule path 'vendor/trezor-crypto': checked out 'f9523f97df2c965835001d551738bbe1b93b53df'
Submodule path 'vendor/trezor-qrenc': checked out '4da44cacb7bd69aa297bafe4a41c498a050225a4'
make: Entering directory '/trezor-mcu/vendor/nanopb/generator/proto'
protoc --python_out=. nanopb.proto
protoc --python_out=. plugin.proto
make: Leaving directory '/trezor-mcu/vendor/nanopb/generator/proto'
make: Entering directory '/trezor-mcu/firmware/protob'
  PROTOC  messages_pb2.py
  PROTOC  messages-bitcoin.pb
  NANOPB  messages-bitcoin.pb.c
Writing to messages-bitcoin.pb.h and messages-bitcoin.pb.c
  PROTOC  messages-common.pb
  NANOPB  messages-common.pb.c
Writing to messages-common.pb.h and messages-common.pb.c
  PROTOC  messages-crypto.pb
  NANOPB  messages-crypto.pb.c
Writing to messages-crypto.pb.h and messages-crypto.pb.c
  PROTOC  messages-debug.pb
  NANOPB  messages-debug.pb.c
Writing to messages-debug.pb.h and messages-debug.pb.c
  PROTOC  messages-ethereum.pb
  NANOPB  messages-ethereum.pb.c
Writing to messages-ethereum.pb.h and messages-ethereum.pb.c
  PROTOC  messages-management.pb
[... 150 lines cut ...]
  DEP     udp.d
  MAKO    nem_mosaics.c
Traceback (most recent call last):
  File "../vendor/trezor-common/tools/cointool.py", line 15, in <module>
    import click
ImportError: No module named 'click'
  MAKO    ethereum_tokens.c
Traceback (most recent call last):
  File "../vendor/trezor-common/tools/cointool.py", line 15, in <module>
    import click
ImportError: No module named 'click'
  MAKO    coin_info.c
Traceback (most recent call last):
  File "../vendor/trezor-common/tools/cointool.py", line 15, in <module>
    import click
ImportError: No module named 'click'
  CC      udp.o
  MAKO    coin_info.h
Traceback (most recent call last):
  File "../vendor/trezor-common/tools/cointool.py", line 15, in <module>
    import click
ImportError: No module named 'click'
Makefile:121: recipe for target 'coin_info.h' failed
make: *** [coin_info.h] Error 1
make: Leaving directory '/trezor-mcu/firmware'

seelezlk, 2018-10-09:
BTW, I have already run the ./build.sh, it seems to work well

prusnak, 2018-10-10:
Quick fix pushed in 7ab36d57abe20a7b2c746e187f5f95378280ef87

@matejcik feel free to fix it more properly (using pipenv?) if you have some spare cycles

PR #419 (2018-10-11): docker: update build scripts and instructions to fix Docker emulator
fixes #417

this removes the separate Dockerfile for emulator and instead adds conditional branching to the main Dockerfile. the same build script is used - `scripts/fullbuild` reacts to the `EMULATOR` env var, and you can specify `EMU` as "bootloader revision" for `build.sh` to build the emulator.
commit b0e879b3 (2018-10-11): docker: update build scripts and instructions to fix Docker emulator (#419)

fixes #417
closed 2018-10-11 by prusnak