trezor/trezor-mcu #420: Get CoinName

MelbourneDeveloper, 2018-10-14:
In older versions of the protobuf definitions there was a CoinType message that had a CoinName. There used to be a list that came with the Features message. It looks like this is gone now. There was also a file in this repo called Coins.json from memory. This also seems to be gone. 

My understanding is that CoinName is still important when calling GetAddress for coins like Bitcoin Cash for example - especially pre-fork addresses. Am I right?

So, is there a list of coin names floating around now? Or, is there some different way we are supposed to use specify the coin to get the correct address? Here is some relevant code in coin_info_c.mako:

https://github.com/trezor/trezor-mcu/blob/555a650eb8bcea352c934ae8d0becd3de80dd58c/firmware/coin_info.c.mako#L25

But, I can't figure out where it is getting its coin data from from...

MelbourneDeveloper, 2018-10-14:
I think I've answered my own question here. This seems to work well.

closed 2018-10-14 by MelbourneDeveloper

prusnak, 2018-10-14:
Correct. There is no way how to get the info from the device and you should look at trezor-common for coin definitions. Also there is a [support.json](https://github.com/trezor/trezor-common/blob/master/defs/support.json) file which declares the minium version which supports such coin.

matejcik, 2018-10-15:
I'll add that you can use `tools/cointool.py` from trezor-common - either the `dump` command that will output the data as JSON, or `render` which will process the Mako templates with the appropriate data fed in