trezor/trezor-mcu #421: Ubuntu 16.04.5 Desktop need to [ sudo apt-get install libsdl2-2.0-0 ]

sarveshwar-singh, 2018-10-19:


prusnak, 2018-10-19:
This doesn't make sense. `./build.sh` builds using Docker and this installs sdl2 into the image.

closed 2018-10-19 by prusnak

sarveshwar-singh, 2018-10-24:
Followed:
https://github.com/trezor/trezor-mcu/blob/master/README.md

default branch is `master` so i was using `master` on a fresh Ubuntu, tried about two times.

For now I am using Trezor T, after Firmware update of Trezor One which is going support Cardano will try to reproduce.

karel-3d, 2018-10-25:
`./build.sh EMU` indeed uses docker, I just tried it

However, at the end, it writes an error `install: cannot stat '/src/build/firmware/firmware/trezor-emulator.bin': No such file or directory`. Not sure if related to the OP.

karel-3d, 2018-10-26:
Oh, but @sarveshwar-singh is right, you still need libsdl2-2.0-0 to RUN the emulator after build