trezor/trezor-mcu #424: install: cannot stat '/src/build/firmware/firmware/trezor-emulator.bin'

karel-3d, 2018-10-26:
Trezor emulator build (build.sh EMU) throws this 

`install: cannot stat '/src/build/firmware/firmware/trezor-emulator.bin'`

Not sure if related to https://github.com/trezor/trezor-mcu/issues/423 ; I am building on macOS

karel-3d, 2018-10-26:
(however it seems that the emulator, even if I manage to build it, is a linux binary, which is useless on macOS, unlike the Trezor T emulator)

karel-3d, 2018-10-26:
Oh it seems the error is not important, since the .elf file is still there.

matejcik, 2018-10-26:
yeah, the error is spurious, i didn't want to spend time on modifying the generic part of the script.

might look into it later

Sjors, 2018-11-29:
Same error for me on macOS and indeed the .elf file does show up.

rafaeltokyo, 2019-02-20:
in https://github.com/trezor/trezor-mcu/blob/master/script/fullbuild#L103 change $EMULATOR_FILENAME to $FIRMWARE_FILENAME, same with EMU  the compile generate trezon.bin file

rafaeltokyo, 2019-02-22:
@karel-3d worked ?