trezor/trezor-mcu #430: when building from docker, the FW cannot be put on the device

karel-3d, 2018-11-06:
When I build the FW by the docker instructions, when trying to upload FW to device, I see

"Unknown bootloader detected.  Unplug your TREZOR contact our support"

I don't know if it's a feature or a bug :(

prusnak, 2018-11-06:
`MEMORY_PROTECT` variable is set to 1 by default, this triggers bootloader check in the FW. If you want to disable the bootloader check - edit the Makefile to set `MEMORY_PROTECT=0`.

However, what you see is a corner case, because a device should always contain a valid bootloader. Maybe you ended up with the strange/custom bootloader in your device after some experiment?

closed 2018-11-06 by prusnak

karel-3d, 2018-11-06:
> However, what you see is a corner case, because a device should always contain a valid bootloader. Maybe you ended up with the strange/custom bootloader in your device after some experiment?

No, I have updated the device to latest FW and BL on beta wallet website. I don't know how I ended up in this corner case.

I am building by tag - `./build.sh 9226589604e78ab3d9f9f90f31bf838e359f34f8`

prusnak, 2018-11-06:
Right, the tag you use does not know the new bootloader (yet). The bootloader you use was added to the whitelist a few commits later - 327aaa99213bc8d079f971bccf6d99c1e7e5b246

karel-3d, 2018-11-06:
Aaah that's it. Thanks!