trezor/trezor-mcu #433: Protection from malicious firmware

eftaxiopoulos, 2018-11-22:
Since when updating the firmware, the bootloader gets updated too, isn't it possible that someone updates my trezor with a malicious firmware and he also updates the bootloader with one that contains the correct public keys in order not to appear the message "unofficial firmware"?

In this case he can install a malicious firmware without me understanding anything?

jhoenicke, 2018-11-22:
The bootloader is updated by the firmware on the first run.  So the old bootloader will check the firmware and will warn you if the firmware is not signed. Only a signed firmware can update the bootloader.   If an unsigned firmware is installed, the bootloader will not only trigger the warning, it will also run the firmware with limited permissions.  In particular, unsigned firmware cannot overwrite the bootloader.

closed 2018-11-22 by jhoenicke

eftaxiopoulos, 2018-11-22:
Thank you very much! In which part of the code the bootloader does not let the firmware make an update? I actually made this question because I have a prototype here (already locked) and I actually want to change my bootloader and firmware with ones that are signed with a different key. Would this be possible? Maybe I could first disable this part of the code that gives limited permissions to unofficial firmware.

saleemrashid, 2018-11-22:
@eftaxiopoulos If you haven't installed the new bootloader, you could install your own firmware that updates the bootloader. You would be able to flash your own bootloader that has your keys instead of (or in addition to) SatoshiLabs' keys.

Then your own firmware would be seen as "official", and would be run with full permissions.

eftaxiopoulos, 2018-11-22:
@saleemrashid I already have my own bootloader that accepts my own firmware. Its accepts my firmware as official. The problem is that now I want to change my keys. I want to sign my firmware with another key, and have a bootloader that accepts this new key and not the old one. But I do not want to brick the device. What is the safest way to do it?

saleemrashid, 2018-11-22:
Are you running the new bootloader that runs unofficial firmware with limited permissions? What method did you use to install this firmware?

eftaxiopoulos, 2018-11-23:
I am running bootloader and firmware compiled by my in a prototype of trezor. The firmware is "official" as it is signed with my key and the bootloader is configure to accept this key (signatures.c).

What actually happened was that I decided that I want to change my private key. Is there any way to do it without bricking my prototype?

In the beginning I thought that any firmware (even unofficial) could update the bootloader and this is why I opened the issue. But as you told me unofficial firmware runs under limited permissions (where is this in the code?). So apparently there is no security issue, but I would still appreciate if I could change my keys without bricking the device.

I thought of the following:

step 1: Install a firmware signed with the old key but putting the new keys in the signatures.c. This will update the bootloader to one that accepts the new keys and after the update the firmware will look like unofficial.

step 2: Install a firmware signed with the new key

saleemrashid, 2018-11-23:
@eftaxiopoulos That is the method I would suggest; just wanted to clarify what you had done so far

> But you told me that unofficial firmware runs under limited permissions (where is this in the code?).

https://github.com/trezor/trezor-mcu/blob/b1725e7264aae72b0ad3c6f4e829f7fc48629a46/util.h#L60-L76

eftaxiopoulos, 2018-11-23:
Thank you very much @saleemrashid !