trezor/trezor-mcu #437: bootloader: return non-empty serial number

prusnak, 2018-12-18:
Reporter from Reddit:

```
It is a problem with the Windows drivers. Device manager says:

Windows has stopped this device because it has reported problems. (Code 43)

The USB device returned an invalid serial number string descriptor.
```

keepkeyjon, 2018-12-18:
Since you can't easily read the one out of storage, what's the best strategy here? I see a few options:

1) Use `desig_get_unique_id_as_string`, the ID baked into the mcu from the factory.
2) Program a new random serial into OTP.
3) Generate a random serial each time the bootloader starts.
4) Lock down the format of the metadata section in storage, and use that.
5) Give the bootloader its own storage sector.

(1) is really bad for privacy. Our device does this currently for both the firmware and bootloader, but I'd prefer it didn't. Firmware will get the fix in our next release, but it'll be a while before we're ready to ship a new bootloader.

(2) is also bad for the same reasons, but less bad.

(3) seems reasonable, but there could be drawbacks I'm not aware of. Do clients rely on the id being stable? Do host OSes care?... Windows is incredibly picky about other usb things, so I wouldn't be surprised.

(4 & 5) seem messy, and restrictive.

prusnak, 2018-12-19:
Best solution is

6) return constant value

see https://github.com/trezor/trezor-core/issues/441

prusnak, 2018-12-19:
Fixed in 2b40fdcdd1f9dc9f6e2dea6ef30bc81b74d5ce75

closed 2018-12-19 by prusnak