trezor/trezor-mcu #440: Compiling firmwares

antonio-fr, 2018-12-22:
I'm trying to compile the firmware binary and to rebuild the correct binary watching the fingerprint.

```bash
#!/bin/bash
# Compile Trezor Firmware from a given commit
rm -Rf trezor-mcu
git clone https://github.com/trezor/trezor-mcu.git
cd trezor-mcu
export LC_ALL=C.UTF-8
export LANG=C.UTF-8
git checkout $1
script/setup
./script/toolchain-download
pipenv install
PATH=$PATH:xxx/vendor/toolchain/bin/
pipenv run ./script/cibuild
```
./buildFW.sh v1.7.3 :
ec4d116f64fb8cd69f64057e9fbc4c80f60061d6b8c30c3280442a8e2c62197c

./buildFW.sh v1.7.2 :
9e05a899af146ea6cc7ee41814b826b9c3ab4c4219fdf8162d9d46e4a4021de8

./buildFW.sh v1.7.1 :
f8907ea19cc08e23841166bfd47eaae86694916111800223849788a07a985a87

./buildFW.sh v1.6.3 :
e8dbb4b7fe8384afd4c99790277c2f2f366a1a0f3957aa3545c75371a99a8fcc

The v1.6.3 is the only one giving the same fingerprint as in data/firmware/1. Any idea why?

prusnak, 2018-12-22:
```
git clone https://github.com/trezor/trezor-mcu.git`
cd trezor-mcu
git checkout TAG
./build.sh TAG
```

Make sure you have Docker installed and running

closed 2018-12-22 by prusnak

antonio-fr, 2018-12-22:
Well, I want to compile without docker.
So do you have any idea what is the root of my issue?
Where can I see the list of the packages, the configuration if the dockers?

antonio-fr, 2018-12-23:
Where can I see the list of the packages / the configuration of the docker?

prusnak, 2018-12-23:
https://github.com/trezor/trezor-mcu/blob/master/Dockerfile

antonio-fr, 2018-12-23:
OK, I get the issue : I replaced the toochain from the downloaded one (script/toolchain-download), to the package of the system and it compiles accordingly the official fingerprints. Mysteriously, the downloaded toolchain compiles the same for v1.6.3 but compiles differently for v1.7.x.

Here's the recipe on a Debian 9, no need of docker :
```bash
apt-get install -y git python3-pip lib32z1 lib32ncurses5 protobuf-compiler python3-protobuf libprotobuf-dev gcc-arm-none-eabi libnewlib-arm-none-eabi
pip3 install pipenv
```

```bash
#!/bin/bash
# Compile Trezor Firmware from a given version
git checkout $1
export LC_ALL=C.UTF-8
export LANG=C.UTF-8
script/setup
pipenv install
pipenv run ./script/cibuild
```

Clean up :
```bash
rm -Rf .local/share/virtualenvs/trezor-mcu*
rm -Rf trezor-mcu
```

saleemrashid, 2018-12-23:
@antonio-fr Presumably, the downloaded toolchain needs to be updated to match Debian 9

antonio-fr, 2018-12-23:
In case the script/toolchain-download script is not used during compilation, better to remove it from the repo?