trezor/trezor-mcu #450: combined.bin size

eftaxiopoulos, 2019-01-29:
I tried to compile the firmware 1.7.3 and I realized that the combined.bin is bigger that 512kB. The output of the prepare.py was:

bootloader : 32768 bytes
firmware   : 489200 bytes
combined   : 554480 bytes

How this is supposed to fit in the 512kB STM32F205RE?

I see that in the fullbuild script the max firmware size was modified from 491520 to:
983296  # 256 + 64*1024 + 3*128*1024 + 4*128*1024

prusnak, 2019-01-29:
Try it :-)

Actually, all F205 models have 1 MB Flash. They just report less.

closed 2019-01-29 by prusnak

eftaxiopoulos, 2019-01-29:
I actually tried and it does not seem to work. When I run:
```
openocd -f interface/stlink-v2.cfg -f target/stm32f2x_stlink.cfg -c "init; reset halt; flash write_image erase PATH_TO_BIN 0x8000000; exit"
```
I get:
```
Open On-Chip Debugger 0.9.0 (2018-01-21-13:43)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
WARNING: target/stm32f2x_stlink.cfg is deprecated, please switch to target/stm32f2x.cfg
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Info : STLINK v2 JTAG v27 API v2 SWIM v6 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.233136
Info : stm32f2x.cpu: hardware has 6 breakpoints, 4 watchpoints
target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0800300e msp: 0x2001fff8
auto erase enabled
Info : device id = 0x200f6411
Info : flash size = 512kbytes
target state: halted
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x20000042 msp: 0x2001fff8
Warn : no flash bank found for address 8080000
wrote 524288 bytes from file
```

It wrote 524288 bytes (=512kB) and not the whole combined.bin size

eftaxiopoulos, 2019-01-29:
So I should install the firmware using the trezorctl for example?

One last question. What is exactly the reason for adding 32768-256 zeros in the combined.bin? Because the actual size of bootloader+firmware could still fit in 512kB. The firmware itself is still smaller than 491520.

prusnak, 2019-01-30:
1) yes
2) the area padded with zeros is used for storage, this needs to be filled in so combined.bin reflects the bytes on the flash