trezor/trezor-mcu #456: Request for arbitrary OP_RETURN support

jcramer, 2019-02-21:
It appears from the referenced code that Trezor only supports a single data push by usage of `op_return_data` field, and also the MCU selects which push code to use in your `op_push()` method based on the size of `op_return_data` provided by API.

Is it possible for Trezor to support arbitrarily formed OP_RETURN outputs? 

What I mean is allowing the user to provide the raw scriptPubKey output to be signed for op_return, which would already contain the 0x6a + valid push codes to use.  The signed output should allow multiple data push operations as required by the user.

Also, would you please confirm that the size of the op_return script is not limited by the hardware's software.

Reference:
https://github.com/trezor/trezor-mcu/blob/19c7c8bc3b76a9489b9a5cce5d9b69d60b1f20cb/firmware/transaction.c#L215

prusnak, 2019-02-21:
OP_RETURN size is limited to 80 bytes as per spec.

Also OP_RETURN is defined as 0x6A opcode + exactly one PUSH (not exceeding 80 bytes).

closed 2019-02-21 by prusnak

jcramer, 2019-02-21:
Hi @prusnak, Thank you for your prompt response.

I believe you are speaking for Bitcoin (BTC)?  

Some other blockchains allow multiple pushes (using any valid push code) and also a larger space for op_return data.  This is why it might be very useful to allow signing with arbitrarily constructed op_returns.

Is there anything that you would have you re-consider this request?

jcramer, 2019-02-21:
@prusnak I think BTC allows multiple data pushes in OP_RETURN:

https://github.com/bitcoin/bitcoin/blob/bccb4d29a8080bf1ecda1fc235415a11d903a680/src/script/standard.cpp#L127

prusnak, 2019-02-21:
Can you point me to specification where it says you can have more pushes than one in OP_RETURN?

jcramer, 2019-02-21:
Here is a transaction created just now on BTC with multiple data pushes:
https://btc.com/706f83685116892f5ce2d10cd2efad080343764109a36160614770c96d370520

You can see an OP_RETURN + 20 x OP_1

Looks like: `OP_RETURN 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1` at the bottom of the linked page.

AlexITC, 2019-03-04:
At Stakenet, we ùse the `OP_RETURN` to create TPoS contracts, we won't be able to do that unless the limit is increased to 140 bytes.

jcramer, 2019-03-04:
@AlexITC 

IMO the MCU code should not be limiting the size of OP_RETURN in the first place because this is network relay policy, not a consensus rule.  It makes sense to handle OP_RETURN specific size validation elsewhere, like by the wallet or just let the network error bubble up to the UI.

Also, blockchains other than BTC have increased OP_RETURN size limits so simplified compatibility across chains is just another reason to leave size validation out of the MCU code.

jcramer, 2019-03-04:
Hi @prusnak based on the information presented above (i.e., new BTC specs etc) Is there any chance you would consider re-opening this issue?

AlexITC, 2019-03-05:
Also, two more quotes from https://en.bitcoin.it/wiki/Script

> Currently it is usually considered non-standard (though valid) for a transaction to have more than one OP_RETURN output or an OP_RETURN output with more than one pushdata op

And

> scriptPubKey: OP_RETURN {zero or more ops}

I have already started doing some analysis on how to support multiple pushes, the `op_return_separators` field could be included on the `TxOutputType` value (https://github.com/trezor/trezor-common/blob/master/protob/messages-bitcoin.proto#L231), with the following rules:
- no bytes means a single push is required.
- otherwise, each byte represent the size for a push, so, first push is at `op_return_data[0]` with length `op_return_data_separators[0]`, second push would be `op_return_data[0 + op_return_data_separators[0]]` with length `op_return_data_separators[1]`, etc.

The protobuf definition could be: `repeated bytes op_return_separators = 10;`

Hopefully, the only place to change the code based on the rules would be at [transaction.c -> compile_output](https://github.com/trezor/trezor-mcu/blob/master/firmware/transaction.c#L215).

matejcik, 2019-03-05:
is there any reason for the "separators" field? as opposed to making the `optional op_return_bytes` into `repeated op_return_bytes`?

AlexITC, 2019-03-05:
just not sure whether it breaks protobuf compatibility, if it doesn't, that's better.