trezor/trezor-mcu #459: Firmware 1.8.0 is not tagged

yura-pakhuchiy, 2019-03-05:
Makes verification by rebuilding and comparing fingerprint not possible by users.

prusnak, 2019-03-06:
It's tagged in the private branch, which will be merged into master today (after the official release).

closed 2019-03-06 by prusnak

yura-pakhuchiy, 2019-03-06:
Fingerprint does not match:
```
$ git status
HEAD detached at v1.8.0
nothing to commit, working tree clean
$ ./build.sh v1.8.0
...

Filename    : /src/build/bootloader-bl1.8.0.bin
Fingerprint : f7fa165be6d780f3e1af00abc07df8b3076bcdad72d70da22a63d8896b6391d8
Size        : 32768 bytes (out of 32768 maximum)

Filename    : /src/build/trezor-v1.8.0.bin
Fingerprint : 09fe5ba04979324e632962553bc3da914b3dfc7385b6bb18cb0b3307c259c5ad
Size        : 519152 bytes (out of 983296 maximum)
```
vs `d65f0c07a6a9c53d8b5287798eb53154b33f9e87cd38a3701970e3d0a750a659` listed in https://github.com/trezor/webwallet-data/blob/master/firmware/1/releases.json

prusnak, 2019-03-06:
We changed the firmware format in bootloader 1.8.0. Your fingerprint `f7fa165be6d780f3e1af00abc07df8b3076bcdad72d70da22a63d8896b6391d8` is the fingerprint of inner data (TRZF firmware header + firmware) used in the new bootloader.

Fingerprint listed in the releases.json is the fingerprint for showing in OLD bootloader, it contains TRZR firmware header + TRZF firmware header + firmware, so it leads indeed to a different fingerprint.

yura-pakhuchiy, 2019-03-06:
Hm, `f7fa..` comes from bootlodader. Have you meant `09fe..`?
If I calculate fingerprint with `sha256sum` on whole file except the first 256 bytes, I'm getting the same `09fe..`. Should not it match what old bootloader will display?
```
$ tail -c +257 trezor-v1.8.0.bin | sha256sum
09fe5ba04979324e632962553bc3da914b3dfc7385b6bb18cb0b3307c259c5ad 
```

yura-pakhuchiy, 2019-03-06:
@prusnak so, is it possible to very fingerprint somehow?
TRZR header from old bootloader is 256 bytes.
If I compare only TRZF + firmware from official firmware:
```
$ tail -c +257 trezor-1.8.0-official.bin | sha256sum
d65f0c07a6a9c53d8b5287798eb53154b33f9e87cd38a3701970e3d0a750a659  -
```
With TRZF + firmware of local built firmware it does not match.
```
$ sha256sum trezor-1.8.0-local.bin 
974d38161e034b25f63a035cb5de4dd224ede5b3a41d968190bcdba6b073510e  trezor-1.8.0-local.bin
```

prusnak, 2019-03-06:
Try the following (this will compare the code portions only, without the headers): 

```
tail -c +1280 trezor-1.8.0-official.bin | sha256sum
tail -c +1024 trezor-1.8.0-local.bin | sha256sum
```

* 1280 = 256 + 1024 (TRZR header + TRZF header)
* 1024 (TRZF header)

yura-pakhuchiy, 2019-03-06:
This one worked. Thanks!
In fact only 192 bytes differs:
```diff
 000001f0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 00000200: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 00000210: 0000 0000 0000 0000 0000 0000 0000 0000  ................
-00000220: 0000 0000 0000 0000 0000 0000 0000 0000  ................
-00000230: 0000 0000 0000 0000 0000 0000 0000 0000  ................
-00000240: 0000 0000 0000 0000 0000 0000 0000 0000  ................
-00000250: 0000 0000 0000 0000 0000 0000 0000 0000  ................
-00000260: 0000 0000 0000 0000 0000 0000 0000 0000  ................
-00000270: 0000 0000 0000 0000 0000 0000 0000 0000  ................
-00000280: 0000 0000 0000 0000 0000 0000 0000 0000  ................
-00000290: 0000 0000 0000 0000 0000 0000 0000 0000  ................
-000002a0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
-000002b0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
-000002c0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
-000002d0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
-000002e0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
+00000220: 9d80 d1af 8e51 1076 3dbf d271 e02e e880  .....Q.v=..q....
+00000230: 08a4 7aa9 7a3a 2fba 3358 0f8b 5d32 27d7  ..z.z:/.3X..]2'.
+00000240: 5ce4 ca76 7987 57fd 0395 7b56 999e 968e  \..vy.W...{V....
+00000250: e704 ed37 28e2 636f db52 7165 2fbb 017c  ...7(.co.Rqe/..|
+00000260: 61e0 c926 d934 aebb 8c3d 470d e675 c8ea  a..&.4...=G..u..
+00000270: beae c046 9dd8 77aa 9093 d398 5bd2 4b51  ...F..w.....[.KQ
+00000280: b7f5 6986 c488 821c 2b3a df26 cda6 655e  ..i.....+:.&..e^
+00000290: eabc a52d 98cb 77ed a74e 9fdf 867c db00  ...-..w..N...|..
+000002a0: c731 9b44 4748 e623 e94d 2b28 d01f 04ef  .1.DGH.#.M+(....
+000002b0: a477 f720 346c 3ff1 654b bd15 e461 be67  .w. 4l?.eK...a.g
+000002c0: e2cb 05f8 3258 c432 1854 95d7 bb8d 2d4e  ....2X.2.T....-N
+000002d0: f61c 2323 c535 7d18 a512 8766 9360 b864  ..##.5}....f.`.d
+000002e0: 0102 0300 0000 0000 0000 0000 0000 0000  ................
 000002f0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 00000300: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 00000310: 0000 0000 0000 0000 0000 0000 0000 0000  ................
```

prusnak, 2019-03-07:
These 192 bytes are 3 ECDSA signatures (3 * 64 = 192)