trezor/trezor-mcu #462: Memory fault after disconnecting T1 during loading (Starting up)

sorooris, 2019-03-11:
T1 fw 1.8.0, no PIN or passphrase set

matejcik, 2019-03-12:
The important feature here is that no PIN is set.

On TT without pin, doing the same thing bumps the failed pin counter. Presumably this is what's happening (and failing) here as well.

When you set a pin, there's no introductory progress bar and disconnecting during powerup is OK. Also, when you enter pin and disconnect when the progress bar is running afterwards, failed pin counter increases but the device works fine.

this is probably for @andrewkozlik

andrewkozlik, 2019-03-13:
This bug appears only on Trezor 1 devices which do not have a PIN configured, i.e. the PIN is set to "". Even when a PIN is not configured, the device still needs to unlock itself using the "" PIN. If this process is interrupted, e.g. by unplugging the cable, then the PIN fail counter gets incremented, as @matejcik correctly noted, and that is what triggers the bug. During the next boot, the device will detect that the PIN is not set and it will try to auto-unlock. Since the PIN fail counter is non-zero, it will first go into the exponential backoff wait loop, which calls hal_delay(), which in turn calls usbSleep(), which calls usbd_poll() and that is what causes the memory fault. All of this happens in config_init(). The problem is that config_init() is called prior to usbInit().

Possible fixes:
1. Initialize the USB prior to config_init(). The problem is that for this we need the USB descriptor, which is in the config, so config_init() needs to be called first :-). There is probably a way to sort this out, but could be messy.
2. If the USB is not initialized, then do not call usbd_dev() in usbSleep(). This seems like the cleanest solution.
3. Disable exponential backoff if the PIN is empty. This shouldn't be a security issue, but I think it doesn't really fix the core of the problem.

prusnak, 2019-03-13:
I vote for option 2.

andrewkozlik, 2019-03-13:
Created pull request https://github.com/trezor/trezor-mcu/pull/464. However, I was unable to test this fix, because I am getting "Hard fault detected" during boot (with or without this commit).

andrewkozlik, 2019-03-14:
Test successful. Fix merged: https://github.com/trezor/trezor-mcu/commit/e1bc0ebc402b2b60ce72b6a774959525bcd0e0f0.

prusnak, 2019-03-15:
Great!

closed 2019-03-15 by prusnak