trezor/trezor-mcu #465: Safety check failed error after upgrading to 1.8.0

rodrigopavezi, 2019-03-15:
Hi There,
I have upgraded Trezor firmware to 1.8.0 but then now I am getting the following error.

[trezor-link] Calling BridgeTransport.call( 382 EthereumSignTx { address_n: [ 2147483692, 2147483708, 2147483648, 0, 0 ],
nonce: '05',
gas_price: '0165a0bc00',
gas_limit: '5208',
to: '23d233933c86f93b74705cf0d236b39f474249f8',
value: '09184e72a000',
chain_id: 3 } false )

Error:

[trezor-link] Done BridgeTransport.call, result { type: 'Failure',
message:
{ code: 'Failure_DataError', message: 'Safety check failed' } }

Does anyone know what could be happening?

Cheers

prusnak, 2019-03-15:
Contact our support at https://trezor.io/troubleshooter

prusnak, 2019-03-15:
Not sure what wallet you are using, but if they are not using Connect 7, tell them to upgrade. This is not a Trezor issue.

closed 2019-03-15 by prusnak

rodrigopavezi, 2019-03-15:
@prusnak I am developer of the SelfKey Wallet which is an electron app. I am not sure Connect 7 will work well with our UX. So we are using trezor.js still. Is there someone from Connect 7 that could help me to understand how I can migrate to it. I need to listen for events like REQUEST_PIN so handling that in our UI then send the PIN to the device callback. I could not find a way of doing that yet.

Cheers

prusnak, 2019-03-15:
Don't use trezor.js directly. Use Connect. If you have trouble with using Connect from Electron, ask here: https://github.com/trezor/connect/issues/245

sk91, 2019-03-15:
@prusnak since electron is not supported yet with trezor connect, is there a way to use trezor.js with this? We have an app in production that depends on this integration.

sk91, 2019-03-15:
please reopen it, it is still an issue

prusnak, 2019-03-15:
It is not a trezor-mcu issue. The issue is in your code.

Please help us get Connect working on Electrum. That's the only way how to resolve this.

tayvano, 2019-03-18:
I left a comment here: https://github.com/trezor/connect/issues/245

Hopefully others in this thread can join the conversation as well on 245. Thanks