trezor/trezor-mcu #468: Hi

mattlaurent, 2019-04-07:


mattlaurent, 2019-04-07:
Hi
I'm trying to implement the TREZOR DIY kit with adjusted firmware for my thesis (on fault injection) and I need to turn off the memory protection and insert some extra commands. If I try to follow the steps from https://mcudev.github.io/trezor-dev-kit I get syntax errors when executing the _make -C firmware/ sign_ command 

`MAKO    nem_mosaics.c
  File "../vendor/trezor-common/tools/cointool.py", line 76
    print(prefix, *args, **kwargs)
                        ^
SyntaxError: invalid syntax`

I don't need this functionality for my purpose but I get these syntax errors at other functions where there should me none. I'm running Ubuntu 18.04.2 LTS with all required packages as described in the tutorial. This happens both when using the toolchain included in the tutorial as well as when the most recent toolchain is used. I also made sure that my computer is set-up according to the steps followed in build.sh (protobuf, python, GNU toolchain,...) but this also doesn't solve the issue.

I also tried building with `MEMORY_PROTECT=0 ./build.sh` and this works aas expected, but in the terminal I see the line MEMORY_PROTECT = 1 popping up during building.

Is it easiest to remove the functions executed when MEMORY_PROTECT is 1 and then use the build.sh (and what would be those functions) or is there another option?

prusnak, 2019-04-07:
```
print(prefix, *args, **kwargs)
^
SyntaxError: invalid syntax
```

My guess is you have python2 as your default python. Try using `export PYTHON=python3` and re-running the build again.

For the second question: try using `export MEMORY_PROTECT=0` and then running the build again.

closed 2019-04-07 by prusnak