trezor/trezor-mcu #469: Omni transaction: wrong amount on Trezor One, correct on Trezor T

dannywillems, 2019-04-08:
Hi,

I've got an issue using the Trezor One to create Tether transactions.

I use Trezor One *and* Trezor T to create and sign tether transactions. When I try on the Trezor T, the USDT amount is correct (5 USDT). When I try on the Trezor One, the amount is not correct. Here two pictures for the same transactions.
The Bitcoin amount is correct, and also the receiving address too.

[image]
[image]

matejcik, 2019-04-08:
please add a log of the parameters that you send to Trezor for signing.

dannywillems, 2019-04-08:
[image]

matejcik, 2019-04-08:
it would be super helpful if you could actually paste the text, so that we don't have to retype it from the picture by hand?

dannywillems, 2019-04-08:
```
{
  "inputs": [
    {
      "address_n": [
        2147483692,
        2147483648,
        2147483649,
        0,
        0
      ],
      "script_type": "SPENDADDRESS",
      "prev_index": 2,
      "prev_hash": "e4d07ba87913e32cbacc5456f70149973dc374efc224ae7db078016af5d17f37"
    },
    {
      "address_n": [
        2147483692,
        2147483648,
        2147483649,
        0,
        0
      ],
      "script_type": "SPENDADDRESS",
      "prev_index": 0,
      "prev_hash": "48bc377a3914ff07af7e3da926053913c0515943171ae0cfc15c943cd75ea15b"
    }
  ],
  "outputs": [
    {
      "amount": "914974.00000000",
      "address_n": [
        2147483692,
        2147483648,
        2147483649,
        0,
        0
      ],
      "script_type": "PAYTOADDRESS"
    },
    {
      "amount": "546.00000000",
      "address": "12eXXt156hevSx14c7pgryQcGLiPBFcPMj",
      "script_type": "PAYTOADDRESS"
    },
    {
      "amount": "0",
      "op_return_data": "6f6d6e69000000000000001f000000001dcd6500",
      "script_type": "PAYTOOPRETURN"
    }
  ]
}
```

matejcik, 2019-04-08:
Thanks. I can reproduce the problem now.

The problem is that the following code is interpreting values as little-endian, whereas TT is using big-endian: https://github.com/trezor/trezor-mcu/blob/master/firmware/layout2.c#L359-L380

given that Omni specification very helpfully _does not specify_ endianness, we'll need to look into the implementation.

dannywillems, 2019-04-08:
`TT` is Trezor T I suppose?
Trezor One uses little-endian and Trezor T big-endian?

prusnak, 2019-04-09:
@matejcik Can you try replacing the `memcpy` call here: https://github.com/trezor/trezor-mcu/blob/master/firmware/layout2.c#L378
with the code that actually reverses endiannes?

Something like this comes to mind:

``` c
(((uint64_t)data[12]) << 56) |
(((uint64_t)data[13]) << 48) |
(((uint64_t)data[14]) << 40) |
(((uint64_t)data[15]) << 32) |
(((uint64_t)data[16]) << 24) |
(((uint64_t)data[17]) << 16) |
(((uint64_t)data[18]) << 8) |
(((uint64_t)data[19])
```

dannywillems, 2019-04-09:
More a question because i'm curious and checking the code too: why does it work on TT and not on T1? do they not use the same representation?

prusnak, 2019-04-09:
@dannywillems there is a bug on T1 which we are fixing at the moment.

dannywillems, 2019-04-09:
Maybe also here must be investigated? https://github.com/trezor/trezor-mcu/blob/master/firmware/transaction.c#L229

dannywillems, 2019-04-09:
> @dannywillems there is a bug on T1 which we are fixing at the moment.

It does not really answer to the question.

matejcik, 2019-04-09:
for the record here is Omni's source confirming that the format is supposed to be big endian: https://github.com/OmniLayer/omnicore/blob/v0.4.0/src/omnicore/tx.cpp#L244-L247

prusnak, 2019-04-09:
@dannywillems the tx does not have to have 20 bytes, there are other methods with different format

matejcik, 2019-04-09:
The person is allowed to sign the unknown transaction in any case.
But @dannywillems has a point - maybe in case the tx is not a "simple send", we should revert to the raw OP_RETURN data being shown. Right now if I send a OMNI operation other than "simple send", no data is shown for it.

prusnak, 2019-04-09:
No